realthunder/FreeCAD_assembly3 #563: sweeped body destorded after transform

mwlandolsi, 2021-04-06:
Hello, When creating a body with sweep and then the body is transformed using the tool transform, the shape of the final body is distorted.

Before :

[image]

After transform :
[image]
[exemple sweep.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6267458/exemple.sweep.FCStd.zip)

realthunder, 2021-04-30:
I think you've attached a different document then the one shown in the screenshot. The problem you have is because you created `Sketch001` using the Sketch workbench button while mapping on some PartDesign feature. This means the sketch is consider an alien feature outside of the body. When the body moves, the sketch will not move with the body hence the error. The correct way to do is to create the sketch using the button in PartDesign workbench as shown below. This will auto add the sketch inside body.

[image]

mwlandolsi, 2021-04-30:
Yes that  is :) I'm sorry. 
When sketch added to the body it was solved

At least, I learned some thing new.
https://user-images.githubusercontent.com/76944804/116690270-0ef25480-a9b1-11eb-8076-c51a15ce07f9.mp4

closed 2021-04-30 by mwlandolsi

mwlandolsi, 2021-05-02:
many thanks @realthunder