realthunder/FreeCAD_assembly3 #564: draft move linked object problem

drom89, 2021-04-10:
Hello Realthunder,

when i try Draft-Move linked object from other file, it dont work is this normal ?  

[image]

realthunder, 2021-04-11:
Can you please provide the sample file shown in the screencast?

drom89, 2021-04-11:
So it seems that it happens only when i link complete Part, if i link only the body its working.
[image]

luzpaz, 2021-04-11:
@drom89 can you please post you Help > About FreeCAD > `Copy to clipboard` info

drom89, 2021-04-11:
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: Czech/Czech Republic (cs_CZ)

luzpaz, 2021-07-17:
@drom89 Thanks + sample file?

luzpaz, 2021-07-30:
@drom89 ping

nadarbreicq, 2021-12-24:
Hi ! Same issue here.

realthunder, 2021-12-24:
Should be fixed by the above commit.

Please note that, you are not supposed to use `Draft_Move` on features inside body. Select the body instead.

closed 2021-12-24 by realthunder