realthunder/FreeCAD_assembly3 #565: Could someone please add a little something to the README?

nrrdzilla, 2021-04-11:
Hi.

This might dovetail with another issue I saw, asking for a stable release, but I was hoping for information that might get some of us rolling well ahead of such a thing coming to pass.

I've been digging around trying to figure out what version of assembly3 is compatible with which builds of FreeCAD. I pretty quickly found a number of references which, unsurprisingly, suggest that the most recent version of assembly3 will only work with the nightly builds of FreeCAD.

But, now that 0.19 is the stable release and has trickled down to the repositories used by the various distros, it would be nice to get assembly3 into there. It's certainly not bundled in the Ubuntu 20.04 release of FreeCAD 0.19, for example. For the life of me I have been unable to find any indication of which assembly3 release would work with these stable releases of FreeCAD. Could someone please point me (and others) to that release?

Please consider two things:
1) a sticky post at the top of this issues page, that answers the above question
2) a note in the README to go read that post, and in addition (so it's hauled down with any git clone and can be viewed offline) the assembly3-release to FreeCAD-release mapping information, which should be a pretty short list

TIA,
n

luzpaz, 2021-07-16:
@nrrdzilla  How does the current [README](https://github.com/realthunder/FreeCAD_assembly3/blob/master/Readme.md) look to you?  
Do you know how to make a Pull Request? You could submit one with your proposed changes. Thoughts?

luzpaz, 2022-07-04:
@realthunder according to the A3 package.xml file, the minimum version of upstream FC 0.19.0 is this correct? If so, lets add that as a prerquisite to the README.