realthunder/FreeCAD_assembly3 #566: New renderer discards model's color

ngocminhdao88, 2021-04-12:
I am trying out the new renderer in lastest daily build. Somehow this new renderer doesn't show my model's colors anymore. Another renderer works fine.

Here is my model:
[workbench-table.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6298009/workbench-table.zip)

Is the file created by daily version binary compatible with stable version?

Here is my FreeCAD info:
> OS: Ubuntu 20.04.2 LTS (i3/i3)
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 2021.412.22506 +5327 (Git) AppImage
> Build type: Release
> Branch: LinkDaily
> Hash: d41e78fe7f9b54eea5977df0a7344331f08445e8
> Python version: 3.8.8
> Qt version: 5.12.9
> Coin version: 4.0.1
> OCC version: 7.4.0
> Locale: English/United States (en_US)

oficinerobotica, 2021-06-06:
Tested with 
```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```
and the issue isn't present any more.

kevenwyld, 2021-06-15:
@oficinerobotica this is still an issue.  I have experienced it with my own model, but for the purpose of this case I also tested with the provided `workbench-table.zip` example.  Simply turning on the experimental renderer and opening the model illustrates the issue quite well.

The standard rendering engine:
[image]

And the experimental one:
[image]

Recreated with:
```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

ngocminhdao88, 2021-06-15:
The new renderer also has the problem with rendering the color from wrl file.

This is the screenshot i took from RobotExample
[image]

Newest FreeCAD version
> OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

oficinerobotica, 2021-06-15:
Indeed I can confirm that with that particular file the colors are lost while using the experimental renderer. I'm curious what this model has different. I can't experience the same problem with any of my files.

```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

kevenwyld, 2021-06-15:
I'm unable to confirm at the moment but with my own model the colors are set on the imported assembly in the included file, not in the main "assembly" file where all the links are. I'll check when I can to see if this is related.

-------- Original Message --------

kevenwyld, 2021-06-15:
OK so I was able to reproduce this with a simple single file assembly.  It seems that the experimental rendering engine does not respect a color if it is set on the assembly it self that contains the part, and not the body.  See the video below.  I've also attached the file.

Video:

https://user-images.githubusercontent.com/11936535/122132180-9d068b00-cdf7-11eb-9163-b62b8a4b517f.mp4

File:
[color_test_2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6659104/color_test_2.zip)