realthunder/FreeCAD_assembly3 #568: Dark Style Sheet data and view tab appearance

vasto7, 2021-04-14:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.24301 +3461 (Git)
Build type: Release
Branch: LinkStage3
Hash: 86b851cf40e8f6999c1b8eec267d5e48207cf376
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
[image]

See image.  When using any dark style sheet, either the data or view tab is white and the text is obsured.

Rabbit-sk, 2021-04-15:
I'm not excited about the current look of the `View` and `Data` tabs either. For example with `Qt 5.12.6` (release 2021.0205) it looked like this:
[image]
After updating to `Qt 5.15.1` (release 2021.0224), the appearance of the `View` and `Data` tabs does not look good:
[image]

It would be good if the appearance of the `View 'and` Data' tabs was the same as before.

vasto7, 2021-04-15:
Yes.  I agree with you.

realthunder, 2021-04-16:
I'm aware of this visual change in Qt 5.15. I'll investigate it later. I'm currently busy preparing the topo naming merge to upstream. So development here will be on hold for a few days.

luzpaz, 2022-01-12:
I have VertUI installed so I can't test ATM :laughing:  
@vasto7 please retest on latest Daily and report back. Thanks!

luzpaz, 2022-06-30:
Can someone confirm on realthunder edge snap ?