realthunder/FreeCAD_assembly3 #569: Feature request - Make manual measuring unnecessary (most of the time)

herrgerd, 2021-04-15:
Hi @realthunder,
today at work I used both Inventor and SolidWorks again. And then it struck me that I don't use the real "measure" function there very often. That has a very simple reason:

[image]
It does show the dimensions of the selected edge, line or face in the lower right corner of the information bar. In this case "Durchmesser" = Diameter: 10mm.

Then I thought: Wait, FreeCAD has this bar too, and it does show some coordinates there.

I looked around a bit and saw that you can select e.g. an edge, hit CTRL+ALT+P to send it to python and get the length by the `elt.length` command.

**Now here's my question:** Would it be much work to add this super helpful feature to FreeCAD? I mean: select an edge, face, or two faces and automatically show the length, area or angle between them.

The most useful things would be:

Edge -> length
circle -> diameter
Face -> area
two faces/planes -> if they're parallel show their distance, else their angle to one another
cylinder -> show diameter

Thanks for all your hard work!!!

Cheers, Jan

realthunder, 2021-04-29:
There are add-on maybe already have this function. See for example the second screencast [here](https://forum.freecadweb.org/viewtopic.php?p=480010#p480010). It shows various geometry information as 3D overlay. I'll ask the poster about which add-on he was using.

herrgerd, 2021-05-04:
> 
> 
> There are add-on maybe already have this function. See for example the second screencast [here](https://forum.freecadweb.org/viewtopic.php?p=480010#p480010). It shows various geometry information as 3D overlay. I'll ask the poster about which add-on he was using.

Hey @realthunder, I tried several add-ons and macros. This is why I asked for the feature to be built-in.
Measuring is one of FreeCADs very few last big issues. Basic measuring operations absolutely have to be accessible without activating a macro or add-on. Measuring distance/angle/length is as important as sketching and constraining...

herrgerd, 2021-05-20:
@realthunder thank you very much for implementing the function!
It is working very well. You got an email from me with a few additional thing I found.

I will close the issue now, I guess if something is buggy, people will open new issues anyway. Cheers, Jan