realthunder/FreeCAD_assembly3 #571: Polilyne in sketch no cycle behaviour with m key.

dgrabulosa, 2021-04-16:
Polilyne in sketch no cycle behaviour with m key.

OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.24301 +3461 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 86b851c
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: Catalan/Spain (ca_ES)
@dgrabulosa

Shkolik, 2021-04-29:
Just tested on latest daily. It start working only on second segment of polyline. Have no idea if it's expected or not, but it is what it is. Please check what shortcut is selected in setting for that command, probably you have to set it by yourself, because it was hardcoded in a past.  I'm using 'X' shortcut for this, btw.

[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.22506 +5327 (Git)
Build type: Release
Branch: LinkDaily
Hash: d41e78fe7f9b54eea5977df0a7344331f08445e8
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

dgrabulosa, 2021-05-01:
It works correctly.
I assign a custom shotcut to the polyline but the tooltip on the icon says cicling behavior with (M) and it confused me.

closed 2021-05-01 by dgrabulosa