realthunder/FreeCAD_assembly3 #572: creating thick body always fails when tapper angle is used

mwlandolsi, 2021-04-17:
hello dear programmers.
When a body/shape is created with "pad" and the option "Tapper angle is enabled, it will be impossible to make a thick body.
[image]
[image]

mwlandolsi, 2021-05-05:
I think that I have find a solution how to choose parameters for thin feature for this issue, but not so easy for simple users. It could be an improvement to get a better UX.

used version : 
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3597 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 540995a6e8d5db01dfe2597439ec3131ca69b030
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

I will show 3 use cases and what kind of parameters should be considered in every case : 
1- just a simple rectangular pad with draft angle , here nothing special to change
[image]

2-a circular pad with tapper angle

 If we proceed like the first case we will get this result : 
[image]

to get a correct result with thick object, first the parameter "Use pipe for draft" should be "true" while creating pad or in "property view" panel

[image]
then it will be possible to get a good result
[image]

3- A rectangular  sketch with fillet between line : 
 first the parameter "Use pipe for draft" should be "true" while creating pad or in "property view" panel
[image]

Here the "joint type" in thickness parameters should be set to "intersection"

[image]

closed 2021-05-05 by mwlandolsi

realthunder, 2021-05-05:
I actually tried to tackle this issue but couldn't make it work. The problem of your opening post only occurs if the second drafted pad touches the edge of the first pad. I think this might be due to some intrinsic limitation of the thickening algorithm in current OCC. There might be other cases that it will fail.