realthunder/FreeCAD_assembly3 #573: Topo naming issue in daily branch 20210412

CADrock1, 2021-04-17:
Hi Realthunder,

Thanks for you contribution in improving the freeCAD. 
I am facing a topo naming error in attached file. After changing the number of lines in first sketch, the child features are not getting updated, seems failed.
I have attached both FreeCAD files before and after changing the sketch.

Thanks & Regards,
CADrock1
[image]
[image]
[topo naming error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6329219/topo.naming.error.zip)

nongkode, 2021-04-24:
another example

[image]

https://user-images.githubusercontent.com/9507464/115954184-c7d01380-a519-11eb-81fa-8188898faf89.mp4

realthunder, 2021-04-26:
Thanks for reporting. One thing you must understand is that the first and most important problem to solve for the new topo name encoding is to be able to detect changes. The secondary feature is to be able to auto deduce what the referenced element has been changed into, and this often requires guess work. Even though the answer may seem obvious to the modeler, the algorithm may not be able to get a definitive answer, so it opt to raise an explicit error, and let user decide.

In both of your examples, the change in sketch causes the face in the pad to change. And the topo naming detects that change, which shows that it indeed does its job, while the upstream will simply map the sketch to an unrelated face, and the error will be shown somewhere else that is often hard to pinpoint.

The old topo name of the face can actually be traced to the same corresponding sketch edge, but that edge may generate more than one faces. The algorithm will do a trace back and find all related faces in the new shape, but it will play safe and only auto change the element reference if it can find a face with the exact same geometry as the original shape.