realthunder/FreeCAD_assembly3 #574: Assembly breaks

GinTonic99, 2021-04-17:
OS: Debian GNU/Linux bullseye/sid (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.22506 +5327 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: d41e78fe7f9b54eea5977df0a7344331f08445e8
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: German/Austria (de_AT)

Hi Realthunder,
While investigating the possibilities regarding parametrized design using configuration tables and sub-shape links I ran into some issues. Attached is a project file to illustrate. This is the concept of the project :

- SS is the master spreadsheet containing the parameters
- each parametrized body  (inside a part for better overview) is parametrized via a configuration table (e.g. SS_Steher) which is referring to the parameters of SS
- for each parameter set a corresponding subshape binder is derived from the master body
- the subshape binders are then linked into the assembly

I observed the followng issues:
1. cascaded spreadheets are not updated: when I change a parameter in SS, the derived values in e.g. SS_Steher are not updated. This has to be done manually. The parameters in SS are maybe only understood by woodworkers. As test case the values marked dark green should be used, e.g. change B2 from 65 to 55 to reduce the height of the frame. The corresponding values for the height in SS_Steher are found in B2-B8. These have to be updated manually.
2. Assembly breaks: after manually updating the spreadsheet, the project can be recomputed by hitting the corresponding button. This breaks the assembly.

Is there anything on user side to be done for improvement?

[Regal_05d.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/6329243/Regal_05d.tar.gz)

Two more questions:
I noted that when an assembly is frozen, the colors diappear. Is this an intended behaviour? If yes may I suggest to change this and to display the colors as set.
I observe strange rendering problems. They do not change when I change the rendering related display settings, see screenshot.
[image]
thank you for reading

realthunder, 2021-04-26:
There is a bug in dependency tracking of referencing aliased cell in external spreadsheet, and is fixed in the above referenced commit.

The assembly constraining element missing problem is tougher to solve, though. Because the mutated shape binder actually contains new geometry that has a mutated history, which can't be used for tracking topo name changes. The recommended way right now is to wrap any mutated shape binder with an assembly, and manually create its constraining elements. Note that the element of an assembly is meant to be used by upper assembly to avoid topo naming problem in the first place.

If you want to get really fancy, you can make an assembly with configuration table, and bind the element link reference as configuration settings. The video below shows how to bind an expression to element LinkedObject. Please note that there is a bug in the current release preventing this from working. Please try that in my next release. Also, you are supposed to link (rather than using shape binder) the sub-assembly, and turn on its `LinkCopyOnChange` for configuration.

https://user-images.githubusercontent.com/1207888/116057636-925e2e00-a6b1-11eb-84da-1a80b8f9e7e0.mp4

PS. I've also fixed frozen shape color problem. Thanks for the report. Next time, please open a new issue for a new problem. It would be easier for me to track.

GinTonic99, 2021-04-27:
Thank you for attending the issues and for your detailed response. I will test it with the next revision and next time open separate issues.

GinTonic99, 2021-05-06:
Hi Realthunder,
many thanks for the updated revision. 

I followed your excellent instructions and adapted my test project. These are my findings:

- a simple project like that one I attached to my first post works perfectly, modified parameters in the master spreadheet are populated and the model updated
- in a more complex project (see attachment) one or two elements occasionally break when a parameter is changed. In order to reproduce one of the parameters marked dark green in spreadsheer SS can be modified. Then normally one or two elements break. It can be repaired within a couple of minutes and then the updates are populated correctly. This is a significant progress.

(I did not yet try your suggestion to make an assembly with configuration table, I need more time for that.)

Below a short macro to replace the linked object by an expression. As I did not find out how to disable the read-only bit from python (I am not a programmer and python is a snake for me), one has to diable the readonly first then select the elements to be trated and strt teh macro. No error checking so I ocnsider it as proof of concept. It worked for my projects but as I do not have the background knowledge I do not know whether it would work anywhere else.

Thank you again for your effort.
[Regal_05h.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/6437071/Regal_05h.tar.gz)

`
import FreeCAD

sel = FreeCADGui.Selection.getSelection()

if len(sel) != 0:
	for i in range (len(sel)):
		o=sel[i]
		str=o.LinkedObject[1]
		j=str.find(".")
		if j > -1:
			str=(str[j+1:])
		o.setExpression ("LinkedObject", "("+o.LinkedObject[0].Name+"._self, <<"+str+">>)")
		`

OS: Debian GNU/Linux bullseye/sid (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.501.24301 +3583 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 478b44886569e8934ee571e285b974a58ba5e890
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: German/Austria (de_AT)

GinTonic99, 2021-05-13:
Somehow I have repaired the model, now it works. At first I tested the model attached above with today's daily revision and saw the same behaviour as before. Then I realized, that the expressions in the linked object fields of elements have disappeared. After generating them again, everything started working as expected. Thank you for your effort, I close the issue and attach for reference the working model.
[Regal_05i.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/6473644/Regal_05i.tar.gz)

OS: Debian GNU/Linux bullseye/sid (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.513.24301 +3631 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 8d53c82eb6383765aca63667e2a4a070a18bdd2d
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: German/Austria (de_AT)

closed 2021-05-13 by GinTonic99