realthunder/FreeCAD_assembly3 #577: Deleting a parameter constraint unlinks parameter constraints farther down the list

brandonbeaumia, 2021-04-18:
[image]

[EMT Adjuster.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6330457/EMT.Adjuster.zip)

My simple rectangle has 4 parameterized constraints linked from a spreadsheet. If the first parameterized constraint in the list of constraints is deleted, the other 3 go from orange to red and are no longer linked to the spreadsheet. If the 2nd one is deleted, the next 2 become unlinked. If the 3rd one is deleted, the final one becomes unlinked.

OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.22506 +5327 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: d41e78fe7f9b54eea5977df0a7344331f08445e8
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

brandonbeaumia, 2021-04-18:
Something else to add to this weird behavior:
[EMT Adjuster Error 2 Example.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6330480/EMT.Adjuster.Error.2.Example.zip)
Try adding a constraint to the bottom horizontal line in Pad -> Sketch005. Any calculable constraint results in 27.8, which is the result of a different constraint already in the sketch. I'm also getting
```
"23:06:53  <Exception> ObjectIdentifier.cpp(698): Array out of bound: 13, 13
in binding 'Constraints[13]'
```
errors while running into these weird constraint dependency problems.

brandonbeaumia, 2021-04-18:
Just confirmed this issue happens in the currently posted 'stable' too:

OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.24301 +3461 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 86b851cf40e8f6999c1b8eec267d5e48207cf376
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

brandonbeaumia, 2021-04-18:
(I just realized I probably should've opened this issue on your FreeCAD fork instead, but I came from the /releases page for the binaries. Let me know if you'd like me to move it over to https://github.com/realthunder/FreeCAD/)

realthunder, 2021-04-18:
Either place is fine with me. But I am currently busy with something else. There will be some delay with the fix. Thanks for the detailed report.

R4DM4N, 2021-04-21:
Awesome. I've run into the same issue today.  It's a real pain. Looking forward to a solution.

brandonbeaumia, 2021-04-25:
I see you've committed a fix, any chance we can get new builds to test? I've been running vanilla FreeCAD for a week as a workaround and it's slowly killing me. =/

realthunder, 2021-04-25:
Sure, I'll make a new release soon, hopefully tomorrow.

brandonbeaumia, 2021-04-29:
Just built from source with a git download from late in the day 04/27 and still having the issue, but perhaps a little less often? It seems harder to pin down. Here is the most recent bit of log before a huge chunk of of these _Array out of bound_ errors started, as well as the actual file I'm trying to complete. [I am aware there may be toponaming errors or other unrelated bugs here, but hopefully this still works to help you debug the sketch constraint problems.]

[Without Tape Clip - Array Out of Bound Errors.zip]

(https://github.com/realthunder/FreeCAD_assembly3/files/6396134/Without.Tape.Clip.-.Array.Out.of.Bound.Errors.zip)

<Sketch> SketchObject.cpp(8795): Invalid constraint expression in Without_Tape_Clip#Sketch011..Constraints[33]: <<calcs>>.hex_cutout_diameter
<App> PropertyLinks.cpp(403): Without_Tape_Clip#Sketch009.ExternalGeometry auto change element reference Without_Tape_Clip#Mirrored001 ;#1c0f;:T3127:5:E.Edge9 -> ;#2c2a;:T3127:5:E.Edge9
<Sketch> SketchObject.cpp(6944): Null shape from geometry reference in Without_Tape_Clip#Sketch009: Mirrored001.;#2c2a;:T3127:5:E
<Sketch> SketchObject.cpp(7514): External geometry Without_Tape_Clip#Sketch009.e1 missing reference: Mirrored001.;#2c2a;:T3127:5:E
<Exception> Missing external geometry reference
Recompute failed! Please check report view.
<App> PropertyLinks.cpp(403): Without_Tape_Clip#Chamfer001.Base auto change element reference Without_Tape_Clip#Chamfer ;#2721;:T3144:5:E.Edge36 -> ;#2d41;:T3144:5:E.Edge36
<App> PropertyLinks.cpp(403): Without_Tape_Clip#Chamfer001.Base auto change element reference Without_Tape_Clip#Chamfer ;#2726;:T3144:5:E.Edge40 -> ;#2d42;:T3144:5:E.Edge40
Recompute failed! Please check report view.
<Exception> ObjectIdentifier.cpp(698): Array out of bound: 34, 15
in binding 'Constraints[34]'

OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27764 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: 8870fb8f85f5738a866708f667da24ead439b33b
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.3.0
Locale: English/United States (en_US)

realthunder, 2021-04-29:
The referenced fix commit is inside LinkDaily branch. I am just about to cherry-pick it to LinkStage3, but haven't done it yet.

brandonbeaumia, 2021-04-29:
Oh jeez. 🤦

Thank you for this. If I get to it before you I'll make sure to pull the daily version down.

brandonbeaumia, 2021-05-14:
Confirmed working as expected now.

closed 2021-05-14 by brandonbeaumia