realthunder/FreeCAD_assembly3 #578: Scrolling 3d view accidentally when using transparent overlay for tree (combo) view

kevenwyld, 2021-04-22:
When using transparent overlay for the combo view (tree view??) pane it is possible (and for me quite common) to start scrolling in the combo view and suddenly be zooming the 3d view without any cursor movement since the cursor will eventually reach an area where there is no text.  I've included a video to demonstrate:

https://user-images.githubusercontent.com/11936535/115643678-75ed7a80-a2da-11eb-8749-165a44c4ec67.mp4

I'm not sure if it's even possible to do anything about this (I have no experience with UI development) but it seems like a good solution would be to only scroll the combo view, or any of the transparent panes, when your cursor is over the entire rectangle of the transparent pane, not just where there is text visible.

Let me know if you'd like more information.

realthunder, 2021-04-23:
I have added setting to add some mouse wheel pass through delay. 

[image]

kevenwyld, 2021-04-23:
@realthunder Thank you, this is very good.  I admit I also did not notice the `Auto mouse wheel pass through` setting which is helpful as well though the timeout is much more natural.  Thanks for your excellent work as always!

closed 2021-04-23 by kevenwyld

realthunder, 2021-04-23:
You're welcome. You might also want to try to add some background to the tree items for better visibility. It's at the top of this UI setting page. Click the color selection dialog, and add some alpha (i.e. opaqueness).

[image]

kevenwyld, 2021-04-23:
Yes that does make things easier to read when the part is behind the tree.  Thanks