realthunder/FreeCAD_assembly3 #582: create mesh causes crash

sslupsky, 2021-04-23:
When creating a mesh, FreeCAD crashes.

OS: macOS 11.2
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.412.22506 +5327 (Git)
Build type: Release
Branch: LinkDaily
Hash: d41e78fe7f9b54eea5977df0a7344331f08445e8
Python version: 3.9.4
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Canada (en_CA)

https://user-images.githubusercontent.com/16888734/115928605-a69cf380-a443-11eb-9953-f6d74948ebeb.mov

[bottom mount.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6367993/bottom.mount.FCStd.zip)

sslupsky, 2021-04-23:
crashes if I attempt o export OBJ file as well.

realthunder, 2021-04-27:
I can't reproduce the problem on my Mac. Can you please try download the MacOS Conda build and see if it still has the same problem? If so, please try with the upstream release too.

sslupsky, 2021-04-29:
I am able to export without crash when using the upstream FreeCAD release.

OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.1)
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

I have not tried the Conda build yet.

sslupsky, 2021-05-23:
Still causing a crash on latest macOS and LinkDaily build.

I can make it crash with a single pad and then select the body and create mesh.

OS: macOS 11.3
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.9.4
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Canada (en_CA)

realthunder, 2021-05-26:
Have you tried my Conda release yet? I only have one Mac with Big Sur, so can't really test this with a non development computer. It is possible that the homebrew bundle missed some library, although I can't identify which one yet.

sslupsky, 2021-06-06:
I downloaded the most recent (2021.531) daily Conda release (20210531) and the crash does not occur.
```
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git)
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: C/Default (C)
```

Using the 2021.531 daily BigSur release, the crash still occurs.

sslupsky, 2021-06-06:
I turned on the log messages and here is a copy of what appears in the report view when I launch the app:
```
10:07:29  OpenGL version is: 2.1 (2.1 ATI-4.4.17)
10:07:29  Run Gui init script
10:07:29  Init: Running FreeCADGuiInit.py start script...
10:07:29  Init:   Searching modules...
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Mesh... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Draft... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/ReverseEngineering... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Drawing... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Test... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Fem... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Spreadsheet... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Web... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/TechDraw... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Path... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/MeshPart... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Show(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Sketcher... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Start... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/OpenSCAD... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Surface... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Robot... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Image... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Inspection... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/asm3(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Idf(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Arch... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/AddonManager... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Points... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Measure(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/PartDesign... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Tux... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Part... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Import... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Material... done
10:07:29  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Raytracing... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/.DS_Store(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/parts_library(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/FCGear(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/render... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/PCB... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/BIM... done
10:07:29  Loading Sketcher module... done
10:07:29  A3 available
10:07:29  Loading Path module... done
10:07:29  Loading PathStock... done
10:07:29  Loading GUI of Part module... done
10:07:29  Loading GUI of Path module... done
10:07:29  Loading PathToolControllerGui... done
10:07:29  Loading PathToolController... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/kicadStepUpMod... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/fasteners... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/CurvedShapes... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/sheetmetal... done
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/Curves(InitGui.py not found)... ignore
10:07:29  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/Assembly4... done
10:07:29  Init: Initializing freecad.asm3
10:07:29  Init: Initializing freecad.asm3... done
10:07:29  Init: Initializing freecad.Curves
10:07:29  Init: Initializing freecad.Curves... done
10:07:29  Init: Initializing freecad.gears
10:07:29  Init: Initializing freecad.gears... done
10:07:29  Init: Initializing freecad.fc_cadquery
10:07:29  Init: No init_gui module found in freecad.fc_cadquery, skipping
10:07:29  Init: Loading FreeCAD GUI
10:07:29  Loading PartDesign module... done
10:07:29  Loading GUI of Sketcher module... done
10:07:29  Loading GUI of PartDesign module... done
10:07:29  Init: Running FreeCADGuiInit.py start script... done
10:07:29  Init: Activating default workbench StartWorkbench
10:07:29  Init: Showing main window
10:07:29  Main window restored
10:07:29  Show main window
10:07:29  Toolbars restored
10:07:29  3Dconnexion driver initialized. Client ID: 57007
10:07:29  Init: Entering event loop
10:07:29  Init: Processing command line files
```

sslupsky, 2021-06-06:
I then captured a copy of the log after creating a pad and selecting create mesh but before I click ok (after I click ok, the app crashes).  In the log "flatmesh" and "pybind11" appear to be missing.
```
10:12:40  OpenGL version is: 2.1 (2.1 ATI-4.4.17)
10:12:40  Run Gui init script
10:12:40  Init: Running FreeCADGuiInit.py start script...
10:12:40  Init:   Searching modules...
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Mesh... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Draft... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/ReverseEngineering... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Drawing... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Test... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Fem... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Spreadsheet... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Web... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/TechDraw... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Path... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/MeshPart... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Show(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Sketcher... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Start... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/OpenSCAD... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Surface... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Robot... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Image... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Inspection... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/asm3(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Idf(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Arch... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/AddonManager... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Points... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Measure(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/PartDesign... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Tux... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Part... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Import... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Material... done
10:12:40  Init:      Initializing /Applications/FreeCAD LS3.app/Contents/Mod/Raytracing... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/.DS_Store(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/parts_library(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/FCGear(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/render... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/PCB... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/BIM... done
10:12:40  Loading Sketcher module... done
10:12:40  A3 available
10:12:40  Loading Path module... done
10:12:40  Loading PathStock... done
10:12:40  Loading GUI of Part module... done
10:12:40  Loading GUI of Path module... done
10:12:40  Loading PathToolControllerGui... done
10:12:40  Loading PathToolController... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/kicadStepUpMod... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/fasteners... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/CurvedShapes... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/sheetmetal... done
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/Curves(InitGui.py not found)... ignore
10:12:40  Init:      Initializing /Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/Assembly4... done
10:12:40  Init: Initializing freecad.asm3
10:12:40  Init: Initializing freecad.asm3... done
10:12:40  Init: Initializing freecad.Curves
10:12:40  Init: Initializing freecad.Curves... done
10:12:40  Init: Initializing freecad.gears
10:12:40  Init: Initializing freecad.gears... done
10:12:40  Init: Initializing freecad.fc_cadquery
10:12:40  Init: No init_gui module found in freecad.fc_cadquery, skipping
10:12:40  Init: Loading FreeCAD GUI
[... cut ...]
49  libqcocoa.dylib               	0x0000000118d89d74 0x118d4e000 + 245108
50  QtCore                        	0x00000001108896b7 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
51  QtCore                        	0x000000011088d852 QCoreApplication::exec() + 130
52  libFreeCADGui.dylib           	0x000000010cb55d79 Gui::Application::runApplication() + 1625
53                                	0x000000010c9f01fb main + 5883
54  libdyld.dylib                 	0x00007fff2042bf3d start + 1
```

sslupsky, 2021-06-06:
I found this discussion on the FreeCAD forum:
https://forum.freecadweb.org/viewtopic.php?p=230306

There is a lot of discussion about the pybind11 dependency.  It seems to me this could warrant a closer look.

sslupsky, 2021-06-06:
It is possible there is a problem with the create mesh from shape process as well.   That is, if MeshFlatteningCommand.py fails, create mesh should bail but looking at the stack trace and code, it appears the process continues which leads to the fatal error.  So, IMHO, two things should be addressed:
1.  The missing pybind11 dependency
2. Error handling during create mesh should prevent create mesh from running if the dependency is missing (and report the error).

sslupsky, 2021-11-02:
Oct 15 macOS build crashes when creating mesh.

```
OS: macOS 11.6
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.6
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Canada (en_CA)
```

luzpaz, 2022-06-30:
@sslupsky is this still a problem on latest Link or upstream stable ?

sslupsky, 2022-07-01:
@luzpaz I have not tried RT's fork in a while now.  Give me some time and I will download and test it again.  The FreeCAD 0.20 weekly from the upstream repo I have been running seems to be ok.  If I recall the build number it was 28909 or something close to that.

luzpaz, 2022-07-01:
@sslupsky can you try the Daily build when you do?

sslupsky, 2022-07-04:
@luzpaz I tried the Daily Big Sur and the Daily Conda (Apr 30, 2022 release) with success on both.

closed 2022-07-04 by sslupsky