realthunder/FreeCAD_assembly3 #583: constraints tool not change on select other tool (20210412 daily)

nongkode, 2021-04-24:
1. Click on **Vertical** Distance constraints icon, now **vertical** distance constraint  tool **active**
2. Click on **Horizontal** Distance Constraints icon, **vertical** distance constraint tool **still active**

Cancel by click other tool group or ESC to deselect

p.s. **console are show click are correctly** but real tool not change

---------------------------------------------------------
Windows 10 2009
FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210412.7z

https://user-images.githubusercontent.com/9507464/115954737-cf44ec00-a51c-11eb-857c-a4e1994a0167.mp4

herrgerd, 2021-04-24:
Can confirm for this Version, too:

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.326.24301 +3437 (Git)
Build type: Release
Branch: LinkStage3
Hash: 3424fe8b5ba5a630622340c952231a1b758e6e4f
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: German/Germany (de_DE)

nongkode, 2021-04-26:
Thank you very much

closed 2021-05-02 by nongkode