realthunder/FreeCAD_assembly3 #584: UX : Sketch constraint does not change when another constraint is choosen

mwlandolsi, 2021-04-24:
Hello everybody;
I do not know if this is a change in the way that user can pick a constraint from the menu bar. But I noticed that when I choose a constraint, for example horizontal dimension, and then I click on vertical constraint button, it still remains on horizontal, and does not change to vertical. I must click on the right button of the mouse or on "esc" key and then pick the other constraint.
This issue did not exist in older releases.

https://user-images.githubusercontent.com/76944804/115955665-d6093a00-a4ef-11eb-9863-be5a2c86cefb.mp4

herrgerd, 2021-04-24:
Yepp, seems to be a bug. See #583

herrgerd, 2021-05-02:
Thanks for fixing @realthunder!

mwlandolsi, 2021-05-02:
thanks to all. the issue is solved.

closed 2021-05-02 by mwlandolsi