realthunder/FreeCAD_assembly3 #586: Need Unicode support for error messages.

Kuzma30, 2021-04-27:
I have assembly with russian names for Parts, Element. When I get assembly3 error messages I see messages like this:
13:21:06  Traceback (most recent call last):
  File "G:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210412\Mod\asm3\freecad\asm3\assembly.py", line 735, in execute
    info = self.getInfo(False)
  File "G:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210412\Mod\asm3\freecad\asm3\assembly.py", line 1163, in getInfo
    return getElementInfo(self.getAssembly().getPartGroup(),
  File "G:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210412\Mod\asm3\freecad\asm3\assembly.py", line 1664, in getElementInfo
    raise RuntimeError('Failed to get geometry element from '
<class 'RuntimeError'>: Failed to get geometry element from ____________Y#Link (Assembly001).1.$Element003.
13:21:06  0.0 <asm3.main> assembly.py(737): Failed to get geometry element from ____________Y#Link (Assembly001).1.$Element004.

" ____________Y#Link (Assembly001).1.$Element003." must be look simular to "ЗащитаY#Link (Assembly001).1.$Element003."
 Is it assembly3 related problem?

realthunder, 2021-04-30:
This is not really a unicode display problem. The document name is internally converted by FreeCAD to make it accessible from Python. The original unicode name is kept in Label. But label is allowed to have duplicates, so it is not the best way to find out which object the message is referring.

There is indeed some problem with asm3 object name printing. I have change it in the commit referenced. It will now print object label before its document name, so instead of ` ____________Y#Link (Assembly001).1.$Element003.`, you will get `  (Assembly001) ____________Y#Link.1.$Element003.` To find out which object it is referring to. Copy the full object path name after the label (i.e. `____________Y#Link.1.$Element003.`). Press `CTRL+F` in tree view, and a search box will appear. Paste the object path name into it, and the tree view will expand and scroll to that object. Press `Enter` to select, or `Esc` to hide the search box.

luzpaz, 2022-01-12:
@ Kuzma30 please retest on latest Daily and report back. Thanks!

luzpaz, 2022-04-08:
@Kuzma30 ping