realthunder/FreeCAD_assembly3 #587: highlight selected body in tree view

bambukouk, 2021-04-29:
using the latest:

`OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
`

I am having problem highlighting selected item in Tree view.
First click in 3D view selects the face and highlights the pad in Tree view
Second click highlights the whole body in 3D view, but in Tree view it still shows just the pad, making it difficult to find in Tree view the body I am selecting in 3D view (unless tree view is expanded).
[image]
[image]
[image]

realthunder, 2021-04-30:
In FreeCAD, each PartDesign feature contains all geometries up to the modeling history of that feature. You can think of `Body` feature as a group that contains those features in the history order. So the behavior you describe is expected. There are two ways to help navigate to the parent body feature. 

One is to enable 'Hierarchy selection' view option as shown below. Once enabled. Repeatedly clicking a geometry element will ascend the selection to its parent item.

[image]

The other prefered way is to use the hierarchy menu, with keyboard shortcut `U, U`. You don't need to select anything, just move your mouse over any geometry element in 3D view, or any item in the tree view, and press that shortcut, a menu will appear showing you all the parents. Left click your mouse to select the item, or, right click to expand its children. Hold `Shift` key and hover over the item to show tooltip for more actions.

bambukouk, 2021-04-30:
Thank you very much indeed, for taking time to help me.
Enabling 'Hierarchy selection' view option has solved it.
As far as I know this was not an option but a default in the main 0.19 or 0.20 versions of FreeCAD?
Hence my confusion.

Keyboard shortcut is very neat as well - thank you!
My only difficulty is remembering all these tricks ...

Thank you for great job you are doing with Assembly 3 👍

closed 2021-04-30 by bambukouk