realthunder/FreeCAD_assembly3 #589: Unknown Command ViewFit

KeithSloan, 2021-05-01:
After I import a GDML file the code looks like

>        if FreeCAD.GuiUp :
>              FreeCADGui.SendMsgToActiveView("ViewFit")

To fit the view in the active window.

With the latest Stable and Daily I get an error message
10:08:42  Unknown view command: ViewFit
And the view remains zoomed in

realthunder, 2021-05-02:
Which branch of your GDML repo did you tried? I didn't get this error.

KeithSloan, 2021-05-02:
As far as I know the the code to send msg to do a ViewFit should be in all branches.

I just tried master and link3 branch and I get the error with latest builds
Error message appears in Report View

> 12:55:05  Open : /Users/keithsloan/MEGA/MEGAsync/CAD Files/GDML/Louis/Luxe_GT_V4.gdml
> 12:55:12  Print Verbose : False
> 12:55:12  Import GDML file : /Users/keithsloan/MEGA/MEGAsync/CAD Files/GDML/Louis/Luxe_GT_V4.gdml
> 12:55:12  ImportGDML Version 0.4
> 12:55:12  Parse : /Users/keithsloan/MEGA/MEGAsync/CAD Files/GDML/Louis/Luxe_GT_V4.gdml
> 12:55:12  running with lxml.etree 
>  12:55:12  Unknown view command: ViewFit
> 12:55:12  End processing GDML file
> 12:55:12  time : 5.0604 seconds

KeithSloan, 2021-05-09:
Seems this is down to a problem on the Mac, when after an import the current view switches to the 'Start' window and as the it is the current active View the command fails. So if the problem on the Mac gets resolved this will then work.

Does raise the question of is there a better way of determining the View that the import relates to, so it could be sent the ViewFit request.

KeithSloan, 2021-05-13:
This seems to be fixed in Daily 05.13 :-)

closed 2021-05-13 by KeithSloan