realthunder/FreeCAD_assembly3 #590: puzzling display issues when assembly frozen

bambukouk, 2021-05-01:
Using:
`OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
`
I am having issues with displaying of frozen assembly.
Pics below (with comments) explain what I am experiencing:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

realthunder, 2021-05-01:
Can you please send me your model file?

bambukouk, 2021-05-01:
Thank you very much!
You have an email with attachment.

realthunder, 2021-05-02:
Just notice that you are still using 0328 release. Please update to the latest 0501 release. The color lost problem has been fixed.

bambukouk, 2021-05-02:
I was just downloading it 👍 
and yes - it works fine in the latest:

`OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.501.24301 +3583 (Git)
`

Big thank you again!

bambukouk, 2021-11-18:
Hi @realthunder 
I am afraid, the issue re-appeared again....
I have an assembly, consisting of a number of sub-assemblies.
Freezing of sub-assemblies is no problem at all, but when I try (after freezing all of the sub-assemblies) to freeze top assembly, it goes grey.
[image]
[image]

I am using latest daily:
`OS: elementary OS 6 Odin (Pantheon/pantheon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
`
I have attached original file (re-named .zip to allow inserting here)
[for_rt.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7564063/for_rt.zip)

bambukouk, 2021-11-25:
@realthunder  
I can confirm (having tested it on other files) that freezing of assembly with frozen sub-assemblies as it's parts leads to this problem :-1: 
You can only have "one level" of freezing if this display issue is to be avoided

bambukouk, 2021-12-08:
Thank you!!
Look forward to all these changes in the next release :+1:

closed 2021-12-08 by bambukouk