realthunder/FreeCAD_assembly3 #591: Unable to select origin point

mwlandolsi, 2021-05-02:
Just create a sketch draw a circle for example and try to put the center of the circle coincident with the origin.
that's all.
[image]

herrgerd, 2021-05-02:
Can confirm this:
[image]

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.501.24301 +3583 (Git)
Build type: Release
Branch: LinkDaily
Hash: 478b44886569e8934ee571e285b974a58ba5e890
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

dingolino, 2021-05-03:
Same here.
If you add for example 2 circles and if you select 1 centre point and move over to the other circle centre point than the origin centre point is "highlighted" but you can mark the centre point of the 2nd circle.

Best regards,

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.501.24301 +3583 (Git)
Build type: Release
Branch: LinkDaily
Hash: 478b44886569e8934ee571e285b974a58ba5e890
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

herrgerd, 2021-05-04:
Hey @realthunder it works again. Thank you!

mwlandolsi, 2021-05-04:
> Hey @realthunder it works again. Thank you!

Haw to get the fixed version ?

herrgerd, 2021-05-04:
> 
> 
> > Hey @realthunder it works again. Thank you!
> 
> Haw to get the fixed version ?

here: https://github.com/realthunder/FreeCAD_assembly3/releases

mwlandolsi, 2021-05-04:
Many thans

> > > Hey @realthunder it works again. Thank you!
> > 
> > 
> > Haw to get the fixed version ?
> 
> here: https://github.com/realthunder/FreeCAD_assembly3/releases

Many thanks

herrgerd, 2021-05-04:
> 
> 
> Many thans
> 
> > > > Hey @realthunder it works again. Thank you!
> > > 
> > > 
> > > Haw to get the fixed version ?
> > 
> > 
> > here: https://github.com/realthunder/FreeCAD_assembly3/releases
> 
> Many thanks

You're welcome! You can close your own threads/issues by pushing the red button left to the green "Comment" button.
That way the issues counter drops by one, which is quite a rewarding thing for @realthunder I think :)

mwlandolsi, 2021-05-04:
Nice job, 
many thanks to all and to @realthunder

closed 2021-05-04 by mwlandolsi