realthunder/FreeCAD_assembly3 #593: Pie menu crash and closes freecad when create new sketch from pie menu icon.

dgrabulosa, 2021-05-02:
When working with custom pie menu the action to create a new sketch from a pie menu icon, sometimes  crash and closes freecad.
How I can report info about these crash?

realthunder, 2021-05-03:
Can you please post here the FreeCAD about information? If you are using Linux or MacOS build, can you also post here the crashed call stack?

dgrabulosa, 2021-05-03:
How I can obtain the crashed call stack??

realthunder, 2021-05-03:
Are you running Linux or MacOS? If it is Linux, please open a terminal, `cd` to the directory that holds the AppImage file, and execute the file. When the program crashes, it will print the call stack trace to the terminal. On MacOS, please open the `Console App`, and find the `Crash report`.

dgrabulosa, 2021-05-04:
/tmp/.mount_freecaxw7KC2
FreeCAD 0.19, Libs: 2021.501R24301 +3500 (Git)
Sheet Metal workbench loaded
double free or corruption (!prev)
/tmp/.mount_freecaxw7KC2/AppRun línia 46: 80784 Avortat                 (s'ha bolcat la memòria) "$PREFIX/bin/$name" "$@"

dgrabulosa, 2021-05-04:
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f8be92f1210]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x18b6a7) [0x7f8be94366a7]
#2  0x7f8beb52ede1 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib64/libFreeCADGui.so+0x4a1
#3  0x7f8be980aa18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Core.so.5+0x6c0
#4  0x7f8bea015928 in QAction::triggered(bool) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Widgets.so.5+0x32
#5  0x7f8bea0172b2 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Widgets.so.5+0xc6
#6  0x7f8bea017a3b in QAction::event(QEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Widgets.so.5+0xb3
#7  0x7f8bea01a7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#8  0x7f8bea02031d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#9  0x7f8beb4de849 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib64/libFreeCADGui.so+0x89
#10  0x7f8be97f4606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Core.so.5+0x86
#11  0x7f8be9bb0b51 in QShortcutMap::dispatchEvent(QKeyEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Gui.so.5+0xd5
#12  0x7f8be9bb0bc2 in QShortcutMap::tryShortcut(QKeyEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Gui.so.5+0x5c
#13  0x7f8be9b88ac0 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Gui.so.5+0xb0
#14  0x7f8be9b96e4c in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Gui.so.5+0x6c
#15  0x7f8be9b99939 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Gui.so.5+0x79
#16  0x7f8be9b86c65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Gui.so.5+0x63
#17  /tmp/.mount_freeca3rqO3I/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f8be0794010]
#18  /tmp/.mount_freeca3rqO3I/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f8be5e9823d]
#19  /tmp/.mount_freeca3rqO3I/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7f8be5e984d1]
#20  /tmp/.mount_freeca3rqO3I/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f8be5e98571]
#21  0x7f8be98230e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Core.so.5+0x5e
#22  0x7f8be97f1019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Core.so.5+0x187
#23  0x7f8be97f4e8f in QCoreApplication::exec() from /tmp/.mount_freeca3rqO3I/usr/bin/../lib/libQt5Core.so.5+0xfd
#24  0x7f8beb45813e in Gui::Application::runApplication() from /tmp/.mount_freeca3rqO3I/usr/bin/../lib64/libFreeCADGui.so+0x40e
#25  /tmp/.mount_freeca3rqO3I/usr/bin/FreeCADLink(+0x44bf) [0x5640713b04bf]
#26  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f8be92d20b3]
#27  /tmp/.mount_freeca3rqO3I/usr/bin/FreeCADLink(+0x47d9) [0x5640713b07d9]

realthunder, 2021-05-05:
I can't reproduce the crash. Are there any pattern for the crash to happen? Can you please capture a screencast video?

dgrabulosa, 2021-05-12:
on terminal at every crash said these:
double free or corruption (!prev)
The only patron y can see : the crash is when enter or leave the sketch edit randomly, not every time, but then starts to crash I can reproduce the sequence to crash many times but if the crash not occurs I can work normally some time more.
Wow I need to make a video of the crash.
There are some script that can start Freecad without any custom personalitation? 
I tree these to start with fresh profile and test whiout custom things.

HOME="$PWD/Settings" FREECAD_USER_HOME="$PWD/Settings" ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210501-glibc2.12-x86_64.AppImage

luzpaz, 2022-07-07:
@dgrabulosa can you still reproduce this problem on latest build ?