realthunder/FreeCAD_assembly3 #595: modelling in Assembly 3 - how to constrain sub shape binder

bambukouk, 2021-05-03:
Not sure whether I should post it here (or on the FreeCAD forum?), but for now here it goes - help, please.

Story board as usual:

[image]
[image]
[image]
[image]
[image]

realthunder, 2021-05-03:
If this binder is for one time use, or in other words, you do not want the binder to follow the bound object's placement, you can change the `BindMode` of the binder to either `Frozen` or `Detached`. If `Detached`, the binder loose the link to the bound object, but will retain its current shape. If `Frozen`, the binder keeps the link, but will not update its shape if the bound object changes. You can manually sync the shape using object context menu in tree view.

bambukouk, 2021-05-03:
Thank you realthunder,
The issue is not that the binder follows the bound object's placement, my concern is that I can independently transform the binder in itself (which would mess the model).
I suppose, you can say  I shouldn't be silly enough to do it...  :-)

mwlandolsi, 2021-05-06:
Could you please create a file that contains a simplified modelling of your requirements ?

bambukouk, 2021-05-06:
You have an email with attached simplified file to show the details.
Thank you.

mwlandolsi, 2021-05-07:
could not download the file! I do not know why. Also it is better to attach it here so that you could find more suggestions.

bambukouk, 2021-05-07:
What do you mean "...could not download the file!..." ?

mwlandolsi, 2021-05-07:
> What do you mean "...could not download the file!..." ?

The file you sent by mail, I click on download button, but nothing happens. I is really strange! I do not why.

mwlandolsi, 2021-05-07:
> Very strange...
> It is only 252 kB attachment and I just sent it again to another gmail
> address and it opens no problem at all.
> Plus I never had any problems doing it in the past?

Hello, after opening the file, sorry could not help
مع تحياتي - Salutations - Mit freundlichen Grüßen
محمد وسام الأندلسي
مهندس آلية
مكون في صناعة البلاستيك
تلفون 27499633 (216+)
M.W. LANDOLSI
Mechanical engineer
Plastics trainer / Formateur en plasturgie
tel (+216) 27499633

realthunder, 2021-05-07:
I still don't quite get what you mean here. I mean you have to explicitly invoke the `Transform` command on the binder, or manually change its `Placement` property to move it. If you are using asm3 mover tool, then it will move the container assembly instead of the binder. Can you please post a screencast showing how you can accidentally move the binder?

bambukouk, 2021-05-08:
Sorry, it is me learning and causing confusion.

Yes, problem is, me explicitely invoking `Transform`  command.
Normally, on fully constrained/locked assembly with 0 DOF:
- Assembly 3 mover tool is greyed out  (if the assembly is toggled active)
- or Assembly 3 mover tool is available (if the assembly is not active) but doesn't have any effect 
- when using `Transform`  command, things just spring back to original position

but for an assembly with sub shape binder weird things happen when I try to use `Transform`  command.

Lesson for me:
avoid using  `Transform`  command on Assembly 3 components at all and use only Assembly 3 mover tools.

Ideally `Transform`  command should not be available/possible for Assembly 3 components? (if it is possible?) to stop dumbos like me trying it :-)

I think we can consider this issue closed?
and sorry again for wasting your time on my "learning" issues.

realthunder, 2021-05-08:
No problem. It is not possible to use the `Transform` command for moving assemblies, because assembly mover is aware of assembly hierarchies and will modify the the `Placement` property of the correct assembly/part object in the hierarchy, while the `Transform` command only works for the selected object.

I think this is the same reason that got you confused. Assembly mover will only modify the `part` object's `Placement`, which may be a sub-assembly, a body, an `App::Part` or whatever. It will never modify any object inside the `part`. Even though it may appear that you are selecting an object inside the `part` (i.e. the binder in your case) when you activate the mover, it will walk up the hierarchy to find the correct `part` object to move.

realthunder, 2021-05-08:
> When you say `It is not possible` do you mean user is prevented from using it? or user shouldn't be using it?

Depending on the user intention. If the user does want to apply some offset to the binder, go for the `Transform` command. For moving part/sub-assembly, use the assembly mover.

As for the recompute failure you encountered. It's hard to say what's the cause. When you change the binder placement, your model actually changes, and its constraining element may change as well.

bambukouk, 2021-05-08:
Understand, thank you!

closed 2021-05-08 by bambukouk

Zolko-123, 2021-05-10:
Hello, I'm also interested in this question:

> Depending on the user intention

I fail to see a use-case where the user would want to move the "binder" independently from the original, as that completely defeats the purpose. Therefore, the behavior of the shapebinder seems much more logical than that of the subshapebinder. But only the latter is App::Link aware.

Zolko-123, 2021-05-12:
> This creates a binder that binds to the `Box` and `Cylinder` inside `Pat001`. This is quite straightforward if you just want to create a binder at the root level. If you want the binder inside some other part, then you'll need to calculate the relative object path to avoid cyclic dependency.

thank-you. I have 2 questions, if you could help in answering them: imagine I have a Part1 in a document, and this part contains a Body1, which has a circular edge Edge1. I want to create a subshapebinder in Part1, linking to the Edge1 in Body1. I have selected Edge1 in the 3D window:

1. How do I get the selected edge ? Probably using getSelectionEx(), but I never really know how to use that one
2. How to create the binder to that Edge1 ?
```
binder = App.ActiveDocument.Part1.newObject('PartDesign::SubShapeBinder, 'Binder')
binder.Support = (App.ActiveDocument.Part1, ['Body1.', 'Edge1.'])
```
?

Thank-you for all your support

realthunder, 2021-05-14:
Normally body display mode is set to `Tip`, so when you select the edge in 3D view, it usually the tip feature of the body being. Let's say the tip is `Pad`. You call `Gui.Selection.getSelection()` with two arguments, The first is the document name, simply pass an empty string for the current document, or `'*'` for all documents. And pass 1 for the second argument, meaning that you want the entire object path, instead of just last object in the path. It will return a list, each entry corresponding to a selected top level object. You get the object path in the entry's `SubElementNames` attribute. The following code converts the return value to a format that can by directly assigned to `binder.Support`. You can select more edges or objects to see the effect.

```
support = [ (sel.Object, sel.SubElementNames) for sel in Gui.Selection.getSelectionEx('', 1) ]
```

HoWilgh, 2021-11-09:
> No problem. It is not possible to use the `Transform` command for moving assemblies, because assembly mover is aware of assembly hierarchies and will modify the the `Placement` property of the [...]

I am still a bit confused.... 
So, we are save as long as we are in AMS3 wb and use only the icon 'Axial move part A-A'  or the shurtcut A-A or right-click on a shape (while in ASM3) ?
And we should NOT use Transform (right-click on any shape) from PartDesign wb?

I worked through this https://bassmatifreecad.github.io/FreecadUserBook/en/asm3/move.htm and do see now the intention and workflow but do not see why I should not use the usual Transform or how I can break things.