realthunder/FreeCAD_assembly3 #596: Datum plane normal to spline on selected vertex not working

mwlandolsi, 2021-05-05:
Hello everybody.
Could any one help me how to create a datum plane normal to curve (spline) at a specific vertex ?
here is the result I could obtain !

https://user-images.githubusercontent.com/76944804/117097391-699a0080-ad63-11eb-9aea-2053295e16fe.mp4

[datum plane on spline.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6425089/datum.plane.on.spline.FCStd.zip)

To open the file just rename and remove the .zip extension from file name.

realthunder, 2021-05-06:
There is a bug in attacher Vertex selection which is fixed by the above commit. You should be able to achieve that by simply select attaching type `Normal to edge`, and then select the edge and vertex.

mwlandolsi, 2021-05-06:
Many thanks,

mwlandolsi, 2021-05-14:
Thank you, the problem is solved.

closed 2021-05-14 by mwlandolsi