realthunder/FreeCAD_assembly3 #597: Loft distorded with small sketch size

mwlandolsi, 2021-05-05:
hell every body.
While trying to design a profile part, (see attached) using the loft function, I got a distotion in the shape of part. this happens with small sketch dimention only. th same chape scaled doe not he a problem.

Normal function
[image]

Distorted function with small dimentions of the shetch.

[image]

here are the files
[Tests and help.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6430562/Tests.and.help.zip)

mwlandolsi, 2021-05-31:
thanks issue disappeared in version 2021.0531

mwlandolsi, 2021-06-01:
[loft with smaller dimention.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6572558/loft.with.smaller.dimention.FCStd.zip)

realthunder, 2021-07-30:
It seems to be another `OCC` issue. The problem seems that OCC section inner and outer circle differently for some reason. I'd recommend to manually create a multi-sectioned circle to match with the other end. Note that there is a problem when sweeping with inner hole, which have just [fixed](#755). It doesn't seem to help with the sectioning problem, though.