realthunder/FreeCAD_assembly3 #598: Object from BREP file will not display (Daily 2021.05.04)

vondav, 2021-05-06:
When I open or import BREP file, object won't display. Same behaviour is also in version 2021.04.11.
STEP file works OK

Steps to reproduce:
1. Create cube in Part workbench and export it as BREP
2. Open or import this BREP file - nothing will display

herrgerd, 2021-05-06:
Can confirm with a part made in PartDesign.

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3597 (Git)
Build type: Release
Branch: LinkDaily
Hash: 540995a6e8d5db01dfe2597439ec3131ca69b030
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

vondav, 2021-05-13:
It is OK in version Daily 2021.05.13
Thanks

closed 2021-05-13 by vondav