realthunder/FreeCAD_assembly3 #600: Issue in DODO workbench in Stable Version (20210405)

CADrock1, 2021-05-06:
Hello Realthunder,

Thanks for the wonderful work.

In DODO workbench, In Frame branch manager the trim tool is not working.

I have attached the FreeCAD file for you reference.  Many Thanks in advance.
[image]

Here is my system information:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3511 (Git)
Build type: Release
Branch: LinkStage3
Hash: b8cd87c34a66a8f7a8177f483cebf79e29273813
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/India (en_IN)

Thanks & Regards,
CADrock1

[DODO WB ERROR.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6436505/DODO.WB.ERROR.zip)

oficinerobotica, 2021-06-04:
@CADrock1 could you please retest with the latest daily and report if the issue is fixed?

CADrock1, 2021-06-04:
Yes, the issue is fixed now.

Many thanks,
CADrock1

closed 2021-06-04 by oficinerobotica