realthunder/FreeCAD_assembly3 #602: Question, does this work only with models created with this software?

berteding, 2021-05-07:
I ask because my model which uses sketches on top of generated surface still brakes completely  if I change a length constraint in a sketch. The model is created with the normal 0.19 version.
Using FreeCAD_0.19.23756-Win-Conda_vc14.x-x86_64

Regards,
Bert (bert.eding_at_gmail.com)
You Know how to replace _at_

realthunder, 2021-05-07:
It should work after you perform a full recompute when you first load the file in my release. By right, a dialog will prompt to do just that when you load the file. Can you please post a file showing exactly how the model breaks?

realthunder, 2021-05-07:
That'll do. If the file contains sensitive information, you can just email it to me. You can either describe the steps you've performed in text, or capture a screencast for the actions.

berteding, 2021-05-07:
Hi Mr Zeng (or how may I call you?)

Ok, no problem and not sensitive.
I will write up what I do this evening and just reply to your email.
By the way I am a software developer myself, I developed CNC control systems (www.edingcnc.com<http://www.edingcnc.com>) was my company before.
I am from the Netherlands.

Regards,
Bert

berteding, 2021-05-07:
Hi Mr. Lei,

I made the description of what I do that corrupts my model, its inside the PDF file.
Then there are 2 files of the model, ToolboxTiger900.FCStd is the original file, ToolboxTiger900_V2.FCStd is the corrupted file where I did as described.
I hope you can identify the problem.

Regards,
Bert

berteding, 2021-05-07:
I made A big mistake, using the wrong version, very stupid, I apologize.
Now I have downloaded the correct version and will try again, I will let you know how it goes.
Regards,
Bert

berteding, 2021-05-07:
Hi Mr Lei,

So please ignore my previous email, this is the right one.
So 2 project files of which the _V2 version is the problem one and without _V2 the original that I want to make shorter.
The pdf file contains my descriptions.

Note that I am a FreeCAD beginner, hopefully you can use my feedback to make the software more stable.

Regards and have a nice weekend,
Bert

realthunder, 2021-05-07:
Hi Bert,

Zheng is my last name. Just call me Lei. Anyway, I can't seem to find any download link in your post. I guess you sent me the files as email attachment which is stripped away by github. Please reply through github web interface and drop your files to the text box for attachment. It likely only accepts certain file types, such as zip.

berteding, 2021-05-08:
Hi Lei,
Now I have attached my files here into a zip file. So you should get it now.
ToolboxTiger900.FCStd is the original project.
ToolboxTiger900_V2.FCStd is the project that is now in error as described in the pdf file.

[Gereedschap-bakje.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6445642/Gereedschap-bakje.zip)

I am curious what you will find.
Have a nice weekend,
Regards,
Bert