realthunder/FreeCAD_assembly3 #603: Crash and strange behavior on tesselation view and shadow view

631557fcd, 2021-05-08:
Using all builds after FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210412-glibc2.12-x86_64.AppImage i get following Issues:

1. Loading projects saved in shadowed view, the shadow is only drawn incomplete and "viewFit" is zooming in very close into the object. It seems as if the renderer is thinking the object is much smaller than it is. To overcome this behavior, i have to load a file saved in "as is" view mode, press "viewFit", go back to the other file, press "viewFit" again. Then the shadow and fitting behavior is back to normal.

2. Using 20210501 release Tesselation View is showing nothing until i select a feature in tree view. When navigating Freecad almost immediatly crashes. Sometimes ealier, sometimes later. Using Version: 2021.501.24301 +3583 LinkDaily, tesselation view shows only a bounding box and crashes immediatly.

`Freecad Info:

OS: Ubuntu 20.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.501.24301 +3583 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 478b44886569e8934ee571e285b974a58ba5e890
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: German/Germany (de_DE)`

`Log File:

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f93b452c210]
#1  0x7f93b3fa6db9 in SoShapeStyleElement::setDrawStyle(SoState*, int) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x9
#2  0x7f93b6a3abb5 in Gui::SoFCUnifiedSelection::Private::applyOverrideMode(SoState*) const from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x175
#3  0x7f93b6a4be33 in Gui::SoFCUnifiedSelection::doAction(SoAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x43
#4  0x7f93b3ec0149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#5  0x7f93b40818a7 in SoChildList::traverseInPath(SoAction*, int, int const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x107
#6  0x7f93b40cf624 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x94
#7  0x7f93b40eb80b in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x2b
#8  0x7f93b3ec0149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#9  0x7f93b3ec0502 in SoAction::apply(SoPath*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x152
#10  0x7f93b6a3b437 in Gui::SoFCUnifiedSelection::Private::setHighlight(SoFullPath*, SoDetail const*, Gui::ViewProviderDocumentObject*, char const*, float, float, float, bool) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x447
#11  0x7f93b6a50143 in Gui::SoFCUnifiedSelection::Private::onPreselectTimer() from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x73
#12  0x7f93b6a50504 in Gui::SoFCUnifiedSelection::Private::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x234
#13  0x7f93b6a50a40 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x50
#14  0x7f93b3ec0149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#15  0x7f93b4081a3a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x16a
#16  0x7f93b40cf5ea in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x5a
#17  0x7f93b40eb80b in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x2b
#18  0x7f93b3ec0149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#19  0x7f93b3ec9e6c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0xac
#20  0x7f93b3ec0e2a in SoAction::apply(SoNode*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x2ea
#21  0x7f93b409aa92 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/../lib/libCoin.so.80+0x72
#22  0x7f93b6adb90e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x1e
#23  0x7f93b6ae3b8b in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x5b
#24  0x7f93b6b26e8a in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0xa
#25  0x7f93b6aecf27 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x17
#26  0x7f93b6af4bd1 in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x301
#27  0x7f93b6aefbb0 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x1a0
#28  0x7f93b6b2be8a in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x8a
#29  0x7f93b6ad4161 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0xd1
#30  0x7f93b4a37019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Core.so.5+0xcd
#31  0x7f93b525f7e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#32  0x7f93b5263f5e in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5+0x70a
#33  0x7f93b6758b79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x89
#34  0x7f93b4a39606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Core.so.5+0x86
#35  0x7f93b52634c4 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5+0x286
#36  /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5(+0x1aefd4) [0x7f93b5299fd4]
#37  /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff7c) [0x7f93b529af7c]
#38  0x7f93b525f7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#39  0x7f93b526531d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#40  0x7f93b6758b79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x89
#41  0x7f93b4a39606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Core.so.5+0x86
#42  0x7f93b4dddec5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Gui.so.5+0x677
#43  0x7f93b4dde91f in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Gui.so.5+0x5f
#44  0x7f93b4dcbc65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Gui.so.5+0x63
#45  /tmp/.mount_FreeCA1sxhc9/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f93abaf5010]
#46  /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f93b10d323d]
#47  /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7f93b10d34d1]
#48  /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f93b10d3571]
#49  0x7f93b4a680e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Core.so.5+0x5e
#50  0x7f93b4a36019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib/libQt5Core.so.5+0x187
[... cut ...] t5Core.so.5+0xfd
#52  0x7f93b66d24ae in Gui::Application::runApplication() from /tmp/.mount_FreeCA1sxhc9/usr/bin/../lib64/libFreeCADGui.so+0x40e
#53  /tmp/.mount_FreeCA1sxhc9/usr/bin/FreeCADLink(+0x44be) [0x5647020814be]
#54  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f93b450d0b3]
#55  /tmp/.mount_FreeCA1sxhc9/usr/bin/FreeCADLink(+0x47d5) [0x5647020817d5]`

Additionally i want to thank you for your work, your branch made me look into Freecad again and now i am actually using it.

herrgerd, 2021-05-08:
> Additionally i want to thank you for your work, your branch made me look into Freecad again and now i am actually using it.

That is **so** true! That's why I support him with a few bucks monthly. If everyone would do it, @realthunder could have somebody working for him I guess ;)

631557fcd, 2021-05-09:
Hello realthunder,

I updated the AppImage a few minutes ago, but the problem ist still there. The bounding box in TesselationView though is gone. The crash appears sometimes earlier, sometimes later. It seems to be coincident with mouse interaction, if the mouse is not moved, there is no crash. I tried recording a screencast. If necessary i could upload the file used in the cast, though the bug appears on every file/new document. 

Again, thank you for your work. I tested Freecad at V0.16 and abandoned it immediatly. By chance i saw a video on youtube about your branch two  weeks ago and gave it a try. I really needed a Linux CAD solution to finally abandon windows completly. And now Freecad is my only CAD. Surly, its not comparable to Solidworks, Spaceclaim and co. But it gets it work done nicely, once you know your way around. Thanks to you and the work of the team of the main fork and a really good community.
[image]

realthunder, 2021-05-09:
Thanks for the appreciation! The referenced commits above only exist in source code. I haven't made a new release yet. I'll do it soon.

closed 2021-06-03 by oficinerobotica