realthunder/FreeCAD_assembly3 #604: Cursor is (nearly) invisible with darker stylesheets

herrgerd, 2021-05-08:
Hi @realthunder,
I really like the general idea of the dark(er) stylesheets. But there is one big issue:

Text and Cursor have nearly identical colors. So, editing fields always has me going cloooose to the monitor the search for the cursor.

[image]

The screenshot makes it look much more prominent than it really is, especially on hiDPI monitors is is nearly impossible to see.
Darker orange is the worst, then green. The other color schemes are okay, but not perfectly visible either.

Could the cursor be 1px or so wider and have more contrast in general? Have a great weekend!

Cheers, Jan

realthunder, 2021-05-09:
There is no easy way to do this using Qt API, which is a shame IMO. Yes, it is possible to do on individual line/text edit controls created by FreeCAD, but there are so many of them created by Qt internals, e.g. the QLineEdit inside the QSpinBox as shown in your screenshot.

If you are on Windows, then you are in luck. You can change the cursor width in `Ease of Access -> Text cursor -> Change text cursor thickness`. You can get even more enhancement if you `Turn on text cursor indicator`.

I have yet to find any viable settings on Linux or MacOS.

herrgerd, 2021-05-10:
> There is no easy way to do this using Qt API, which is a shame IMO. Yes, it is possible to do on individual line/text edit controls created by FreeCAD, but there are so many of them created by Qt [...]

Thank you for that tip. Didn't even know I could change that in W10, because I neder hat the need to do so.
I made a short screencast for you, first three examples are the three "darker" themes. The last one is the "darf orange".

With that I wanted to show you what I wanted to show in the first place:

[image]

It's more the contrast between text/cursor-color and background of the field than anything else. So, that should be easily adjustable, right?

herrgerd, 2021-05-15:
Hi @realthunder,
thank you for adding the cursor width to the UI options.

Unfortunately it makes things worse with the stylesheets in the above GIF.

I think I wasn't precise enough in what I said: It is the cursors **color** which makes all the difference. Again, see the gif above. 

- Black text and black cursor with the orange theme: perfectly visible.
- white text and white cursor with the blue theme: perfectly visible.
- transparent grey cursor with the green scheme: which cursor?!

realthunder, 2021-05-15:
The text width option should have no effect on color. So at least it can't be worse, no? Here is a long [post](https://stackoverflow.com/a/55889624) explaining the difficulty of setting text cursor color. In short, Qt 5.8 onwards will force showing text cursor as inverse color of the content. So I think a wider blinking cursor will definitely make things more obvious. It is still not so visible on darker green if the cursor is not blinking over any text, because the green background is too dark, making the inverse not so bright as well. You can either opt to other scheme, or manually change the stylesheets (which is not easy I admit, but gives you much more freedom). You can find the stylesheets [here](https://github.com/realthunder/FreeCAD/tree/LinkStage3/src/Gui/Stylesheets), just copy and rename it and put it in `c:\Users\<username>\AppData\Roaming\FreeCAD\Gui\Stylesheets\`.

herrgerd, 2021-05-15:
> The text width option should have no effect on color. So at least it can't be worse, no? Here is a long [post](https://stackoverflow.com/a/55889624) explaining the difficulty of setting text cursor [...]

I will try that! Loaded it in VScode and it does look very much like CSS.

oficinerobotica, 2021-06-06:
@herrgerd given the explanation form RealThunder would you consider this issue kind of solved? if so it could be closed.
@realthunder when I first fired up the latest daily I really thought that the wider text cursor is a glitch of how the appimage renders qt elements. This is while using my usual link.user.cfg . Can you please reset the default width of the cursor while leaving the customizing options for whoever needs them?

realthunder, 2021-06-06:
I've already changed back the default value. But the change won't apply to existing LinkDaily users unless they manually clearing their link.user.cfg file.

herrgerd, 2021-06-06:
> @herrgerd given the explanation form RealThunder would you consider this issue kind of solved? if so it could be closed.
> @realthunder when I first fired up the latest daily I really thought that the [...]

Hi, unfortunately I don't think this fix did better the situation. The problem with the wider cursor is that it does not push the character next to it further away, but it "encloses" it. This doesn't help visibility in any way.
I think this is especially a problem with high DPI screens like the one I'm using.

I added a full-width GIF to show what I mean:
[image]

The wider cursor (5px) just "eats" the char next to it...