realthunder/FreeCAD_assembly3 #605: Separate pick radius for points vs lines

jpedrick, 2021-05-09:
On the latest daily release 2021-05-04, picking points has become very difficult in sketch mode. I have to zoom in until the display accuracy offsets the points from the line. Using gg doesn't work if you need to pick multiple points, as the previous selection is un-selected. I think the issue could be fixed, or at least worked around, if lines had a different pick radius vs vertexes.

bambukouk, 2021-05-09:
Just had exactly the same problem, I am afraid.
Closed model in Link, opened it in 0.20 and picking points in sketch edit worked perfectly OK.

`OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3597 (Git)
Build type: Release
Branch: LinkDaily
Hash: 540995a6e8d5db01dfe2597439ec3131ca69b030
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
`

krkpatrck, 2021-05-09:
I can not select a vertex in sketcher. i had a sketch i padded then went back to make and edit. i deleted a line and tried to make a new one. however, i was never able to attach the line vertex to the previous vertex. i could highlight the line but never the vertex.

realthunder, 2021-05-09:
Can any of you provide a screencast recording for the problem, starting from creating the sketch and then show the highlighting problem? I still can't reproduce the problem even with @bambukouk `link.user.cfg` file.

krkpatrck, 2021-05-10:
ill be back on tomorrow to see if i cant get a screenshot of the issue.

jpedrick, 2021-05-10:
I'm not that savvy with screencasting, but I played around with a model until I started having the vertex selection problem. What it seemed like I had to do was close the sketch and return to it. 

Other things I did first: 

- spin the object around and sketch on the back
- offset the sketch by -5mm inside the object

@realthunder maybe if you open this and try editing Sketch002 you'll be able to replicate the issue.

[vertex_selection_problem.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6449532/vertex_selection_problem.zip)

bambukouk, 2021-05-10:
@jpedrick 
> I'm not that savvy with screencasting,

I have played with your file and the problem is exactly the same as I am experiencing.
I have attached screen grab (hopefully self explanatory)
[2021-05-10 09-59-10.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6450904/2021-05-10.09-59-10.zip)

herrgerd, 2021-05-10:
I can confirm the behaviour!
It occurs only, when you have an sketch with a negative offset. Then you can't select vertex points and it doesn't connect points.

[image]

krkpatrck, 2021-05-11:
ok so..... while im in sketcher. If i spin my sketch not a full 180 as to edit it from the back but more like a 45 I can then grab vertexes.

realthunder, 2021-05-11:
Finally, this problem should be fixed now. Thanks all for reporting and the follow up. As a bonus, I have also enhanced datum label in sketcher so that it will always has a readable orientation no matter how you rotate the view, or even editing through a link that mirrors the sketch.

bambukouk, 2021-05-11:
That's excellent news (about editing thorugh a mirrored link) - lot of my stuff is mirrored, so really appreciate it.

jpedrick, 2021-05-12:
@realthunder thanks for the quick fix!

herrgerd, 2021-05-13:
Yay, the picking problem with negative offsets seems to be gone now!

@realthunder Unfortunately there's a **new bug** now: The vertical dimensions in sketches are mirrored now:
[image]

realthunder, 2021-05-13:
@mwlandolsi, the turning green thing is not really a bug. It is simply showing your sketch as fully constrained, because all your points are fixed by the coincidence constraint to the external vertices.

mwlandolsi, 2021-05-13:
@realthunder so why this does not happen in the second time, I don't think so. You can see the video after the second 13. Yes all points are green, but I have put only one point and the others are putted without user intervention !

oficinerobotica, 2021-06-03:
@jpedrick Closing this issue as it really seems to be resolved although we don't have different pick radius for point and line. If you think otherwise it can always be reopened. Cheers

closed 2021-06-03 by oficinerobotica