realthunder/FreeCAD_assembly3 #607: ask the author of the AppImage to change the Categories in their .desktop file for better menu placement

snelweg, 2021-05-09:
Appimages work great, but I wonder why FreeCAD is always inserted into the 'other' menu on my Debian Duster install. Can I move it to a more appropriate place like 'development' or 'graphics'?

[image]

I asked at the Appimage repo, they said:

`Not an AppImage issue. You need to ask the author of that AppImage to change the Categories in their .desktop file.
`
https://github.com/AppImage/AppImageKit/issues/1133 # 

So do you want to change it to a  more suiting menu category?

realthunder, 2021-05-13:
Hmm... wonder if there is any way for user to set the category. I mean it's obviously not possible to choose a category that satisfy every one.

snelweg, 2021-05-13:
> Hmm... wonder if there is any way for user to set the category. I mean it's obviously not possible to choose a category that satisfy every one.

No, according to the AppImage documentation the category is set when assembling the Image itself. I do agree it would be best if people could choose themselves.

QuakeString, 2021-05-22:
I think "Graphics" could be the best category for FreeCAD

realthunder, 2021-05-23:
FreeCAD.desktop list the category as `Engineering`, which is quite appropriate. So the problem boils down to how to add that category to your Linux distro.

snelweg, 2021-05-23:
> FreeCAD.desktop list the category as `Engineering`, which is quite appropriate. So the problem boils down to how to add that category to your Linux distro.

Ok, will add that category to my menu's. Thanks for looking in to this.

luzpaz, 2022-07-04:
Should we notate this somewhere ? Please advise

snelweg, 2022-07-05:
> Should we notate this somewhere ? Please advise

What does that mean?

luzpaz, 2022-07-05:
Add this to documentation so users can be aware of this possibility.