realthunder/FreeCAD_assembly3 #611: Draft move bodies issue

mwlandolsi, 2021-05-12:
Hello,
I faced so many problems with use of move tool of the draft workbench, especially with part-design created bodies.
Just create bodies and move them using draft WB

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3597 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 540995a6e8d5db01dfe2597439ec3131ca69b030
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

Please see the link of this video

https://youtu.be/Axt3DQve3Og

herrgerd, 2021-05-13:
I must admit, you're using FreeCAD in ways I would never think of ;)
Might be a bug though...

mwlandolsi, 2021-05-13:
> I must admit, you're using FreeCAD in ways I would never think of ;)
> Might be a bug though...

what is strange ? :)

mwlandolsi, 2021-05-14:
It looks that solved in 20210513 version.
Thanks

closed 2021-05-14 by mwlandolsi