realthunder/FreeCAD_assembly3 #612: Vertical dimensions mirrored in sketch

herrgerd, 2021-05-13:
Hi @realthunder,
I thought I create a new issue for this instead of spamming another thread with it:

Vertical dimensions are mirrored when first created:
[image]

When you rotate the sketch to the back (bottom on the nav cube), it still is mirrored, but when you move the sketch a few pixels (MLB), it is correct again:

[image]

As you can see it changes while draggin the sketch, too.

Cheers, Jan

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.513.24301 +3631 (Git)
Build type: Release
Branch: LinkDaily
Hash: 8d53c82eb6383765aca63667e2a4a070a18bdd2d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

realthunder, 2021-05-13:
Gee! How did I miss that...

herrgerd, 2021-05-20:
> 
> 
> Gee! How did I miss that...

Is is not mirrored anymore, but still need to be rotated 180° after creation :)
Vertical dimensions should always be rotated CCW, like here: https://knowledge.autodesk.com/support/autocad-for-mac/learn-explore/caas/CloudHelp/cloudhelp/2021/ENU/AutoCAD-MAC-Core/files/GUID-F856E617-EF26-4FD0-BA71-3E573130BE46-htm.html

55 is wrong:
[image]

55 is right:
[image]

Cheers, Jan

oficinerobotica, 2021-06-04:
the issue isn't present anymore testing with: 
```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

closed 2021-06-04 by oficinerobotica