realthunder/FreeCAD_assembly3 #613: Experimental renderer renders parts transparent

herrgerd, 2021-05-13:
Hi @realthunder,
maybe this is a known behaviour, but I guess it should be the same between different renderers.

Experimental:
[image]

"Auto":
[image]

Is this a bug?

realthunder, 2021-05-13:
Can you please share the model file? BTW, I see you have `Use OpenGL VBO` disabled. You probably want to enable this option to see huge improvement in graphics performance for large models.

herrgerd, 2021-05-15:
> 
> 
> Can you please share the model file? BTW, I see you have `Use OpenGL VBO` disabled. You probably want to enable this option to see huge improvement in graphics performance for large models.

I had to "re-create" the situation and found that it only happens when I re-use geometry to have two (or more) overlaying additive pipe features, see sketch attached.

Maybe this is not the nicest way to do things, but I like it because I can use one sketch without any projected external geometry or binders etc.

[addPipeTransparent.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6485438/addPipeTransparent.zip)

Cheers, Jan

herrgerd, 2021-05-20:
@realthunder does seem to be fixed in 5.19. Thank you!

closed 2021-05-20 by herrgerd