realthunder/FreeCAD_assembly3 #614: Rename a label (in tree view) takes two attempts

bambukouk, 2021-05-13:
At first I thought it was a problem with my mouse (and my twitchy fingers) but exactly the same behaviour happens when I use F2 (with mouse left well alone) to rename a label in tree view.
First attempt select "rename" ends up with nothing and I have to re-select a cell and "rename" it again (second time always works OK).

It could be my hardware, it could be me, so in the first instance can someone, please (particularly using Win10) confirm this as a real issue?

It's not just the latest release (it has been happening for a while) but I hesitated reporting it, in case it is something to do with me...

`OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.513.24301 +3535 (Git)
Build type: Release
Branch: LinkStage3
Hash: f09c99a04eeb116b45328953020d226adb924992
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
`

herrgerd, 2021-05-13:
[image]

Works for me with F2 and Rename in the DropDown.
Single clicking does nothing though.

bambukouk, 2021-05-13:
Thank you @herrgerd 

What do you use to create .gif? and cropped as you did?
I use OBS studio and they don't do .gif (as far as I know)
I would like to share with you what is happening to me ...

herrgerd, 2021-05-13:
> 
> 
> Thank you @herrgerd
> 
> What do you use to create .gif? and cropped as you did?
> I use OBS studio and they don't do .gif (as far as I know)
> I would like to share with you what is happening to me ...

I use Screen to Gif: https://www.screentogif.com/
I love that tool. Super quick and easy. It's free, too!

bambukouk, 2021-05-13:
Excelent recommendation!
Thank you for that.

As for my re-naming issue... it has mysteriously dissapeared for the moment.
Totally puzzling, apart from the fact that this seems to confirm that it is not hardware or user issue (neither has changed since I have reported it). 
So I will have to monitor it and come back when/if I catch it again... 
and see whether I can guess what it seems to be connected to?

bambukouk, 2021-05-16:
Captured an example of what happens to me.
Here, I have just created new element and selected it (to rename), 
clicked F2, first time and it just "clicks out", then second time round it works fine... :-(
It happens with monotonous regularity...

[image]

bambukouk, 2021-06-29:
Now that I have moved to linux:
`OS: elementary OS 5.1.7 Hera (Pantheon/pantheon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United Kingdom (en_GB)
`

**_the problem is gone._**
I went back to Win 10 and it has started again immediately.
So, I guess it is not FreeCAD, but OS specific issue?
I am closing this issue, because it is OK in linux, and that's OK for me.

luzpaz, 2021-06-29:
If the issue persists in Windows, we should still track this then...right?

bambukouk, 2021-06-29:
Yes, indeed
but... nobody else confirmed it, so I have assumed that it was just me :-)

luzpaz, 2021-06-30:
We need a windows user to test this

bambukouk, 2021-11-30:
after a lot more testing I am reasonably confident that this is nothing to do with FreeCAD, simply because similar things happened in other apps. Solved by changing the mouse ...

closed 2021-11-30 by bambukouk

kevenwyld, 2021-11-30:
@bambukouk I didn't see this until now that you have closed it but I have experienced this issue but with slightly different specifics:

* Only in the Assembly3 workbench
* Only when using the F2 shortcut (though to be honest I don't normally use the mouse to rename, so maybe just observation bias)

It's as if the window focus is changing immediately after using the F2 shortcut.  Were you able to figure out what external issue was causing it?  It's really frustrating but as you said, doesn't seem to be related to FreeCAD.

Some possible external factors I've thought about but have been unable to confirm:

* I use a 3dconnexion space mouse pro
* I use the i3wm window manager in Linux which sometimes does weird stuff with window focus if the size of the window changes or if a new window is opened and then closed again
* My keyboard requires pressing a modifier to allow access to the F* keys.  Sometimes releasing this modifier creates a keyboard event which is visible in xev

bambukouk, 2021-11-30:
@kevenwyld 
yes, my experience was similar to yours.
Difficult to say if is assembly3 only, since I use assembly3 95% of time ... but I guess you might be onto something?
I use el cheapo Kensigton (wired) mouse. I have found that wired tended to be more reliable and better for CAD (YMMV)
Using keyboard shortcut didn't make nay difference (i.e. was just as bad), so maybe it is not the mouse?
It is frustrating and difficult to investigate and solve?
I have been using both Windows and Linux, so I don't think it is anything OS related?