realthunder/FreeCAD_assembly3 #616: "close" from edit mode not work often. ( single click on "close" button has no action)

QuakeString, 2021-05-13:
Refer to [the video](https://drive.google.com/file/d/1_wC3odwbxPnAqz1S_l0YTyOXiKJCDa0e/view?usp=sharing), sketcher or any edit mode can't close using "close" button, some times double-click on "close" button works, some time not. Single click on "close" button has no action.
Here is FreeCAD detail
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.513.24301 +3535 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f09c99a04eeb116b45328953020d226adb924992
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
[image]

realthunder, 2021-05-14:
I can't reproduce the problem. Is this happen every time, or is there any triggering condition?

QuakeString, 2021-05-14:
> I can't reproduce the problem. Is this happen every time, or is there any triggering condition?

Yes, Every single, It only trigger if I double-click multiple time.
[here is another example with different model](https://drive.google.com/file/d/1YzWplXTFc7b7xIGY5rDJ6Z2TPw99S9hD/view?usp=sharing)