realthunder/FreeCAD_assembly3 #619: (linkdaily) Property view becomes invisible when hovered over

jdblack, 2021-05-15:
Hi Guys,

I've been having some difficult using link daily over the last couple of months due to  the font colors changing whenever I hover over  the views tabs on the left.    Clicking on a view causes the  font and background to turn to white on white, which makes it touch to use.

I suspect this is happening because I use the dark theme on mac. Is there a way to disable dynamic color changing for the view menus on the left? 

[image]

OS: macOS 11.2
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.513.24301 +3631 (Git)
Build type: Release
Branch: LinkDaily
Hash: 8d53c82eb6383765aca63667e2a4a070a18bdd2d
Python version: 3.9.4
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

[image]
[image]

realthunder, 2021-05-15:
Please change the `Overlay stylesheet` to dark theme. By default, it is set to `Auto`, and will auto choose a dark theme if the application theme contains the keyword `Dark`. Maybe you are using your own theme without the keyword? Anyway, you might want to try the `Dark-Outline` theme, and change some treeview settings to make it more visually appealing as shown below.

[image]
[image]

jdblack, 2021-05-18:
Oh thank you!  Changing overlay style sheet alone made a huge difference!

closed 2021-05-18 by jdblack