realthunder/FreeCAD_assembly3 #624: Unexpected result with sweep

mwlandolsi, 2021-05-19:
Hello dear Freecaders,
When trying to sweep an arc throught a B-spline, I got an unexpected result when the endpoints of the guide path are closer. the error does not held when they are opener.

this is the result : 

[image]

this is the sketch of the path

[image]

When the guide is opener at the endpoints ther is no problem :

[image]
[image]

mwlandolsi, 2021-05-19:
[test unexpexted sweep.FCStd.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/6507273/test.unexpexted.sweep.FCStd.tar.gz)
this is the file

realthunder, 2021-07-30:
I got the same result in upstream, so I am labeling this as an `OCC` issue. I played around a bit. It seems that if you anchor the arc to the other end of the path, it works. Not sure why. I've checked vertex index. The one you anchored should be the right start. Another observation. If you orient the arc the other way, then it works.