realthunder/FreeCAD_assembly3 #625: Part design: Hole thread clearance does not work directly from input dialogue.

Miksu80, 2021-05-20:
For some reason when using threads, you can't modify it's clearance as it is greyed out.

[image]

When entering from parameters, it works:
[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

bambukouk, 2021-05-20:
Checked 0.20 and can confirm that it is OK there, so definitely Link branch only.

herrgerd, 2021-05-20:
Can confirm for:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

Miksu80, 2021-05-23:
@realthunder can I just close this when you have done a commit that references this? I'm quite new on git and don't know how these work :)

realthunder, 2021-05-26:
Sure, you can close if you think it is resolved.

Miksu80, 2021-05-26:
Should be ok

closed 2021-05-26 by Miksu80