realthunder/FreeCAD_assembly3 #626: could not select the origin point in sketch

mwlandolsi, 2021-05-20:
Hello,
this issue is observed on both 2021.05.13 and 2021.05.19
just create a rectangle for example an place the first point on x axis and the second in the other side of the Y axis. the absolute origin should be between tow points. as in the image bellow.

[image]
Just rotate the 3D view from rear view for example, and you can select the origin.

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3554 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: dc52bafc6f0f34d269972419335705e23c6dbefa
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

herrgerd, 2021-05-20:
I can confirm this.
You can't select the root point when a line is going straight through it:
[image]

In the example GIF I wanted to make the lines endpoints symmetric to the Y axis. This does not work even when using the
`G G` command with which you can select the root point.

@realthunder Unfortunately `G G` is of very limited - if no - use at the moment. It makes me able to pick points, lines and constraints, but I can't do anything with them, once selected (other than deleting). Could this be improved?

bambukouk, 2021-05-20:
Can you select it using selection box?

closed 2021-05-28 by mwlandolsi