realthunder/FreeCAD_assembly3 #627: Part workbench: Fillet fails / not generating correctly with drafted walls

Miksu80, 2021-05-21:
When doing drafted walls and adding fillets to them, they're not correctly generated:
[image]
[FilletedDraftedWalls.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6521117/FilletedDraftedWalls.zip)

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

herrgerd, 2021-05-22:
Can confirm.
Sometimes joining with `Refine TRUE` helps to avoid those situations, but this seems to be a bug indeed.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

mwlandolsi, 2021-05-23:
[image]
confirmed, but I think that this new function of tapper angle is still experimental. Not only this bug happens with it. I encourage the developers an @realthunder for their efforts. I think that they are working on it.

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3554 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: dc52bafc6f0f34d269972419335705e23c6dbefa
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

Miksu80, 2021-05-23:
Yep, tested with cuts and it works, don't know that does it fail with drafted walls..

[image]

 I know that bug squashing is a basic part of sw development -> I've been squashing bugs with commercial cads also. Realthunder and other active members do a superb job and I enjoy using FreeCAD :)

Miksu80, 2021-05-24:
Just tested 2nd fillets around the part in previous post and it fails:

[image]

I've had the same error also when using mirror with complicated part but didn't make an issue because of it was not very simple shape.

[FilletedElement.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6529704/FilletedElement.zip)

closed 2021-05-28 by Miksu80