realthunder/FreeCAD_assembly3 #628: Experimental render cache: Sections do not work

Miksu80, 2021-05-21:
Either in sketch mode or trying to do the clipping plane, when on experimental render cache, it doesn't slice the view. Basically it doesnt do anything with experimental. When auto render cache, all works fine.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

herrgerd, 2021-05-22:
Can confirm.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

Miksu80, 2021-05-28:
Appears to be fixed

closed 2021-05-28 by Miksu80