realthunder/FreeCAD_assembly3 #629: mouse cursor

bambukouk, 2021-05-21:
In the latest version, mouse cursor has changed from the one I am used to (i.e. simple line):
[image]

to black box:
[image]

OK, it still works, but it is driving me nuts :-)
How can I, please, go back to the usual?

`OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
`

realthunder, 2021-05-21:
You can set the text cursor width in `Preference -> Display -> UI -> General -> Text cursor width`.

bambukouk, 2021-05-22:
Thank you, and sorry to have bothered you with such trivia :-)
I tought, since I didn't (knowingly/consciously) change anything, it might have been something you did in the latest release...

closed 2021-05-22 by bambukouk