realthunder/FreeCAD_assembly3 #630: How do I constrain solid models to skeleton sketch (not an issue - question, please)

bambukouk, 2021-05-22:
I would like to try using skeleton sketch:
https://github.com/realthunder/FreeCAD_assembly3/wiki/Create-Skeleton-Sketch
to assemble and simulate a motion of an assembly.

One question which is not clear to me, is:
how do you constrain solid bodies to the skeleton sketch?

snelweg, 2021-05-22:
You can use a shapebinder or use the

bambukouk, 2021-05-22:
Thank you @snelweg 
So, what you are saying is that having created `skeleton sketch` in Draft (as described in the link in my post) 
you use it to create solids bound to it by shape binder or similar?

My problem is that I want to try to add `skeleton sketch` to an existing design to animate mechanism.
I was hoping that there was a way to constrain existing solids to a skeleton sketch, 
as described in the original tutorial:
https://solvespace.com/linkage.pl

> Once the skeleton of the linkage exists, solid models may be constrained against it. 
This is how the picture at the beginning of this tutorial was generated. 
If those constraints are chosen appropriately, then the solid models will follow the skeleton of the linkage 
as the linkage is worked. etc, etc ...

I was looking for a way to do it in Assembly 3.

snelweg, 2021-05-22:
I stopped using draft when I started using RT's linkstage version with it much improved part design workbench. So I can not help you with that, part design is what I use. I use skeleton sketches all the time and they work great in many situations.

Looking at the date of the example you mentioned I think it is best to invest time and effort in methods that are up to date. Everything I see in the example is easily made in Part Design, works great with Asm3

bambukouk, 2021-05-22:
I have never used Draft and didn't plan on using it, but this is what @realthunder is using for animation of linkage system,
and this is what I am interested in, so I follow his wiki and his tutorial.

Why is he using it? if as you say:
> Everything I see in the example is easily made in Part Design, works great with Asm3

Do you know something that he doesn't?
or should this wiki be removed and replaced?

realthunder, 2021-05-22:
Assuming you created the `solid` yourself, then you must be using some sketch for that. If so, simply wrap the `solid` together with the sketch into a sub-assembly, and create some `Element` to one of the sketch element, which can then be used in upper assembly. If the `solid` is imported, or none of your own sketch contains useable element for constraining, then you can create a sketch specifically for that purpose, and wrap it together with `solid` into the sub-assembly.

The key concept of `asm3` is to make a part (or component, or sub-assembly, whatever) and declare its `connector` interface using `Element`, so that upper assembly can know how to connect the part with others.

You can find an example usage of skeleton sketch [here](https://github.com/realthunder/FreeCAD_assembly3/issues/53#issuecomment-419344910). Note that you don't have to use Draft for skeleton sketch as shown in the tutorial. A normal sketch will work just fine, maybe better if it is strictly for 2D, as you can use the sketcher's solver which works better in 2D.

bambukouk, 2021-05-23:
> simply wrap the solid together with the sketch into a sub-assembly

That's excellent @realthunder -thank you!

> you can create a sketch specifically for that purpose

I guess that for my design (where I use both the original `solid` and it's `link(mirror)` in the sub-assembly)
I will have to use additional sketches for each, 
because using original sketch form the `solid` for `link(mirror)` would cause me problems?
Anyway... I will experiment and try -easy enough.

> Note that you don't have to use Draft for skeleton sketch as shown in the tutorial

Perhaps it should be made clear in the wiki?
I got the impression form the wiki that one had to use (for some reason) Draft...
It's good news for me because I much prefer Part Design sketch :-)

bambukouk, 2021-05-24:
Just a final not of thank you to @realthunder 
All has worked very well and not very difficult at all :-)
Below and example of the assembly - just one LH side so far, adding RH will be no problem at all.
I have used separate sketches for each side.
The only thing I had to do differently to what you describe at #53 was:
I couldn't  use construction mode for circles in skeleton sketch, because I had no way of selecting them on the screen?, but since I didn't use these sketches to create any solids this was not a problem.
I think this "issue" can be safely closed.
[image]

closed 2021-05-24 by bambukouk

realthunder, 2021-05-25:
> I couldn't use construction mode for circles in skeleton sketch, because I had no way of selecting them on the screen?, but since I didn't use these sketches to create any solids this was not a problem.

There are a few ways to create element on sketch internal geometry (i.e. external, construction, etc.). The simplest is to drag and drop while editing the sketch. You can also use sketch export. Or create an empty sub shape binder, select the geometry while editing sketch, and drop to the shape binder to reference it.

[image]

bambukouk, 2021-05-25:
> There are a few ways to create element on sketch internal geometry (i.e. external, construction, etc.). The simplest is to drag and drop while editing the sketch. 

Thank you,
clearly missed the obvious 😀
I thought you couldn't create element while in sketch edit mode...

Solved!