realthunder/FreeCAD_assembly3 #631: new export as child option shows duplicate sketches when sketch group is enabled

kevenwyld, 2021-05-24:
I don't know if this is intentional or not but when exporting geometry from a sketch as a child object the child sketch is shown twice in the tree if `auto auxiliary group` is enabled.  Here's a screen shot for clarity.  See how `inlet-trace-sketch-outer` and `Export001` are shown twice in the tree, once as a child of their parent and once as it's own object in the Sketches group.

[image]

realthunder, 2021-05-25:
It's kind of intentional. The purpose of aux grouping is for easy finding of different type of auxiliary features. The `Sketches` group is for 2d objects, and `SketchExport` is a type of 2d object, so it is grouped there. The aux grouping has no effect on other object's grouping behavior, so the export is still grouped by its owner sketch.

kevenwyld, 2021-05-26:
Ah, thanks for the explanation.  Sounds like there's nothing to do then so I'll close this.

closed 2021-05-26 by kevenwyld