realthunder/FreeCAD_assembly3 #632: Linear pattern : could not select edge as a direction reference 20210519 and 20210513

mwlandolsi, 2021-05-25:
hello, 
while trying to create a linear pattern of a pocket (for example) in a direction of an edge, the reference edge could not be selected.
I remember that this was possible to do in older version (may be in Mars 2021)

[image]

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3554 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: dc52bafc6f0f34d269972419335705e23c6dbefa
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-05-26:
The reason why you can't select because what you are seeing is the pattern feature itself, and of course a feature can't reference itself. This problem has been solved in my `Daily` branch by supporting preview mode when editing pattern feature. I have just picked this change to the `Stable` branch here. The screencast below shows that when the preview mode is active, the opaque object you are seeing when editing is the feature being transformed, and the transparent ones are the new patterns. 

https://user-images.githubusercontent.com/1207888/119621203-6e207900-be38-11eb-927c-69f8f43946b4.mp4

herrgerd, 2021-05-26:
> 
> 
> The reason why you can't select because what you are seeing is the pattern feature itself, and of course a feature can't reference itself. This problem has been solved in my `Daily` branch by supporting preview mode when editing pattern feature. I have just picked this change to the `Stable` branch here. The screencast below shows that when the preview mode is active, the opaque object you are seeing when editing is the feature being transformed, and the transparent ones are the new patterns.
> Peek.2021-05-26.15-34.mp4

Very nice! That's a good improvement. Though I generally would advise to use datum/master axis/planes for patterns. Much more robust against changes....

**By the way:** Would it be possible to implement what is standard in any CAD software out there?
Selecting how many features you want and the distance between them?
In the original FreeCAD forum many asked for this over the years, but it seems the devs don't think this is a useful feature...

What do you think @realthunder ?

Cheers, Jan

mwlandolsi, 2021-05-26:
Thanks for your emprovement dear Zheng Lei @realthunder . But I did not noticed this problem with circular pattern when it "references itself".

closed 2021-05-31 by mwlandolsi