realthunder/FreeCAD_assembly3 #636: Closing document with links

Shkolik, 2021-05-27:
Found weird behavior - when trying to close document with links to another documents freecad start opening linked documents. It's little bit annoying if project big enough. Force to close Freecad completely and open again to get rid of all documents.

My opinion, when document wasn't opened intentionally there is no reason to open it automatically.

[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

oficinerobotica, 2021-06-06:
@Shkolik  Version: 2021.531.24301 +3702 (Git) AppImage using File -> Close All will close all documents leading you to the Start Page.  I see an issue tough  if you have a unrelated file opened that you want to keep opened.

Shkolik, 2021-07-13:
> @Shkolik Version: 2021.531.24301 +3702 (Git) AppImage using File -> Close All will close all documents leading you to the Start Page. I see an issue tough if you have a unrelated file opened that you want to keep opened.

This workaround doesn't work on complex assemblies on Windows. See related issue #701 

I even don't able to close all documents one by one if some linked docs in assembly were not opened explicitly. I mean I can close and looks like there is no single doc opened, but when closing FreeCAD I definitely will get a message about illegal storage access. A bit annoying, like a fly in a room... you can live with it, but cannot be happy :)