realthunder/FreeCAD_assembly3 #637: Pocket selected face - resulting shape not refined

Shkolik, 2021-05-27:
When pocketing using face resulting shape appears not refined, but after applying any "genuine" PD feature any residue vertexes/edges disappear as expected. 

[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

realthunder, 2021-05-31:
This could be some tolerance issue. After pocketing, the face is entirely flat, so it can't be merged. Can you please post the file showing the problem? I'll check.

Shkolik, 2021-05-31:
"After pocketing, the face is entirely flat, so it can't be merged. "

Yep, and if any other feature will be applied later this extra edges will gone.

[PocketFaceRefine.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6568083/PocketFaceRefine.zip)

oficinerobotica, 2021-06-17:
@realthunder I tested the example file by first deleting the "Fillet" and "Pocket001" then re doing the pocket and the issue is solved. The shape is correctly refined.
[image]

What i noticed tough is that, with this particular file, on first opening my personal display options aren't respected. Notice the thickness of the lines in the screenshot above and the screenshot bellow. The bellow one is on first opening.
[image]

I also get  artifacts while moving the 3dview. I get the same behaviour with both Auto and Experimental renderer.

```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

realthunder, 2021-06-17:
> I also get artifacts while moving the 3dview. I get the same behaviour with both Auto and Experimental renderer.

What artifacts do you get? Maybe a screencast?

> What i noticed tough is that, with this particular file, on first opening my personal display options aren't respected. Notice the thickness of the lines in the screenshot above and the screenshot bellow. The bellow one is on first opening.

This is normal. The line width is set per view object. The author of the original file has a different default value configured which are used by all object he created. When you create the new object it uses your default value.

oficinerobotica, 2021-06-17:
@realthunder  I understand now about the line width. Something new learned. For some weird reason I was getting a strange flikker on lines when rotating the view that I can't reproduce anymore after a pc reboot. Closing this as the main issue is solved. Thank you

closed 2021-06-17 by oficinerobotica