realthunder/FreeCAD_assembly3 #638: Polar pattern of pocket + fillet

Shkolik, 2021-05-27:
Polar pattern doesn't applied for fillet, only for pocket - see screen recording.

[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git)
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

mwlandolsi, 2021-05-27:
Hello, I confirm !

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3554 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: dc52bafc6f0f34d269972419335705e23c6dbefa
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-05-31:
I can't reproduce the problem. Can you please post a file here showing the problem?

mwlandolsi, 2021-05-31:
Hello,
thanks for your answer, here is what i got : 
[polar pattern with fillet.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6567475/polar.pattern.with.fillet.FCStd.zip)

Shkolik, 2021-05-31:
My example as well, hope it will help:
[PocketFilletPolar.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6568086/PocketFilletPolar.zip)

realthunder, 2021-06-02:
I've fixed this problem. But the fix maybe disruptive, so it will only be available in the next LinkDaily release. Once the fix is well tested, I'll merge it to LinkStage3 release.

Shkolik, 2021-06-20:
Tested on LinkDaily 0614 Windows. Works like a charm! Thank a lot!

closed 2021-06-20 by Shkolik