realthunder/FreeCAD_assembly3 #640: Freecad crash at starting

domomib, 2021-05-29:
After download and uncompress and install freecad asm daily OSX conda Py3 Qt5 20210528 x86_64.dmg.
Launche the software freecad it crashed with the following reasons:

Dyld Error Message:
  Library not loaded: @rpath/libexpat.1.dylib
  Referenced from: /Users/USER/Desktop/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib
  Reason: Incompatible library version: libCoin.80.dylib requires version 10.0.0 or later, but libexpat.1.dylib provides version 9.0.0
My hardware imac, OS: Mac OS High Sierra 10.13.6

slaarti, 2021-05-29:
Can confirm this crash, with the same Dyld error message, with both stable 20210528 and daily 20210528-2 OSX-Conda-Py3-Qt5. I'm on a late 2013 27" iMac running Catalina 10.15.7.

Good thing I kept my 20210519 DMGs to roll back to, I suppose...

thomasgi1, 2021-05-30:
As a work-around you can install the packet "expat" using home brew and copy and symlink the home brew version into the application image:
`brew install expat`
`cp /usr/local/Cellar/expat/2.4.1/lib/libexpat.1.8.1.dylib /Applications/FreeCAD.app/Contents/Resources/lib`
`cd /Applications/FreeCAD.app/Contents/Resources/lib`
`ln -sf libexpat.1.8.1.dylib libexpat.dylib`
`ln -sf libexpat.1.8.1.dylib libexpat.1.dylib`

Samantha-uk, 2021-05-30:
Fab!  I was having the same problem and tried doing some brew magic which did not work.
Having applied what you said @thomasgi1 OS: macOS 10.15 Version: 2021.528.24301 +3686 is now working fine!
Thank you.

realthunder, 2021-05-31:
Just made a new Daily release with corrected libexpat. Not sure if it works. Please try.

thomasgi1, 2021-05-31:
The problem is fixed in new Daily release 20210531.

Thank you.

domomib, 2021-05-31:
Hi, I just start Freecad now with  the new Daily release 20210531 on my Mac.
It took a long time (1 or 2 min) first time to start but it running
It' a good job 
Thank you.

closed 2021-05-31 by domomib